nRF24/RF24 #239: RPi2 w/ Ubuntu ARM: undefined reference when trying to build examples_linux

hy565, 2016-04-04:
Hi guys,

I'm trying to get the library running on a Raspberry Pi 2, but I'm having some problems. I think it could be because I'm running Ubuntu 14.04 LTS on my RPi2 instead of Raspbian. 

I git cloned the RF24 library and had no problems installing it using "sudo make install".

However, when I tried to build the examples to test out the hardware, I'm getting a whole bunch of "undefined reference" errors, shown in the screenshot below:

[image]

The installation seems to install the library files to /usr/local/include and /usr/local/lib, I don't know if that's right...but it seems that the compiler can't find the appropriate header files.

Has anybody encountered something similar or can suggest any way I can troubleshoot this?

Thank you.

mz-fuzzy, 2016-04-04:
Hi @hy565 
This is result of my last commit, sorry :-)
I'll be able to test/detect problems on ubuntu later today.
Until then, to unblock you, undo my latest change by:
`git reset --hard a1d346a9d18d318da8ee0638969f12f064b0351f`

hy565, 2016-04-05:
Hi @mz-fuzzy,

Thanks for the quick reply. I don't think it has to do with your last commit though. I tried making the examples_RPi folder before the weekend (before your commit) and faced the same error. I saw your new commit and tried it out again with the examples_linux folder and hit the same wall.

I've also did the git reset and tried to make the examples_RPi folder again, and sure enough, I am still getting the same undefined reference errors.

I'm still trying to figure out what exactly is the difference in my Ubuntu ARM setup that prevents the example from compiling, and I hope you or anyone here can help me out!

Thanks.

mz-fuzzy, 2016-04-05:
fixed by 061341ed5945d7808fd82ab840c6d673bee12389
@hy565 please check.

hy565, 2016-04-05:
Your latest commit fixed it, I am able to compile the examples on Ubuntu now. 

Thanks @mz-fuzzy !

> ubuntu@ubuntu:~/RF24/examples_linux$ make
>arm-linux-gnueabihf-g++ -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib gettingstarted.cpp -lrf24 -o gettingstarted
>arm-linux-gnueabihf-g++ -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib gettingstarted_call_response.cpp -lrf24 -o gettingstarted_call_response
>arm-linux-gnueabihf-g++ -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib transfer.cpp -lrf24 -o transfer
>arm-linux-gnueabihf-g++ -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib pingpair_dyn.cpp -lrf24 -o pingpair_dyn
>pingpair_dyn.cpp: In function 'int main(int, char**)':
>pingpair_dyn.cpp:201:42: warning: 'len' may be used uninitialized in this function [-Wmaybe-uninitialized]
       radio.write( receive_payload, len );
                                          ^
>ubuntu@ubuntu:~/RF24/examples_linux$

>

closed 2016-04-05 by hy565