nRF24/RF24 #241: Is '2 pin mode using time-division duplexed SPI' possible?

yvesgoeleven, 2016-04-09:
I'm using the nRF24l01+ module with Attiny85 in 3 pin mode today, and am wondering if it is possible to  use the library in 2 pin mode as described by Ralph Lancaster here: http://nerdralph.blogspot.be/2015/05/nrf24l01-control-with-2-mcu-pins-using.html

Basically he's using only 1 pin to support both MOSI and MISO, by alternating the I/O direction during send/receive.

Is this setup possible today? If yes, how do I enable it? And if not, what would it take to make it possible? (Willing to lend a hand if needed)

TMRh20, 2016-04-10:
Probably possible, but it hasn't been coded or tested in this lib. I guess *somebody* would need to either incorporate the code into RF24 (needs code changes + testing) or roll your own compatible code using the provided functions.

yvesgoeleven, 2016-04-10:
I'll give it a try... any specific things to keep in mind when doing dev/test of this library?

TMRh20, 2016-04-10:
Heh, mainly just *does it work?* The main functionality is demonstrated in the examples, via gettingstarted, gettingstarted_callResponse and pingpair_dyn examples, so that is a good place to start when looking at testing functionality after changes. 

Also ATTiny support is kind of 'built in' to the main library files, but this might be easier by following the [portability template](http://tmrh20.github.io/RF24/Portability.html) for SPI. Maybe, I haven't looked at it closely.

closed 2016-11-10 by TMRh20