nRF24/RF24 #243: watchdog and radio.powerDown() don't seem to work in pingpair_sleepy.ino

gitgrimbo, 2016-04-12:
Hi,

If I modify the [pingpair_sleepy.ino](https://github.com/TMRh20/RF24/blob/v1.1.6/examples/pingpair_sleepy/pingpair_sleepy.ino) sketch to always be the `ping_out` role, the `do_sleep()` and `radio.powerDown()` calls don't seem to work.

I have a Pro Mini (16MHz) and it's always using around 31mA (powered by 5.17V).  The Pro Mini by itself draws 19mA when active and 3.4mA when in power down state.

The picture below shows the Pro Mini connected.  I'm using an [nRF24 socket](http://www.ebay.co.uk/itm/191762714694?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT) which itself draws 3.9mA (for its LED and voltage regulator) and this [nRF24L01](http://www.ebay.co.uk/itm/191784627035?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT) (pictured below).

I've exaggerated [the delay here](https://github.com/TMRh20/RF24/blob/v1.1.6/examples/pingpair_sleepy/pingpair_sleepy.ino#L153) to 5 seconds so I can more clearly see what the current draw is.

I only see the 5 seconds delay in the serial console, I don't see an extra 8 seconds for the watchdog timeout (I have set the watchdog delay to 8 seconds `setup_watchdog(wdt_8s)`).

[image]

[image] 

[image]

gitgrimbo, 2016-04-12:
Update.  If I use the [LowPower library](https://github.com/rocketscream/Low-Power) and remove all watchdog related code, and use the following as the sleep code, the Pro Mini drops to 17mA usage.

````c
    LowPower.powerDown(SLEEP_4S, ADC_OFF, BOD_OFF);
````

TMRh20, 2016-04-12:
A couple things to try/consider:

- See [this](https://github.com/TMRh20/RF24/issues/195#issuecomment-173091940) chart
- Some modules won't power down, try using stopListening() as an alternative
- If the IRQ pin from the radio is not connected to pin 2 of Arduino (IRQ 0) add ``` pinMode(2,INPUT_PULLUP); ``` to your code, else sleep will be interrupted

gitgrimbo, 2016-04-13:
Hi,

I tried the following test, and I can't get the Pro Mini under 16.5mA using the following methods (the current draw is in the comments).

`LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);` brings the current usage from 28mA down to 16.5mA, but I can't seem to get the radio to go to a lower power state now matter what I try.

````c
// https://github.com/rocketscream/Low-Power
#include <LowPower.h>
// https://github.com/TMRh20/RF24
#include <RF24.h>

#include <SPI.h>
#include <printf.h>

// Set up nRF24L01 radio on SPI bus plus pins 7 & 8
RF24 radio(7, 8);

byte addresses[][6] = { "1Node", "2Node" };

void setup()
{
	Serial.begin(115200);

	radio.begin();
	radio.setPALevel(RF24_PA_MAX);
	radio.setDataRate(RF24_2MBPS);
	radio.openWritingPipe(addresses[1]);
	radio.openReadingPipe(1, addresses[0]);
}

void loop()
{
// 28mA
	Serial.println("8s Normal delay");
	delay(50);
	delay(8000);

// 16.5mA
	Serial.println("8s LowPower.powerDown");
	delay(50);
	LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);

// 28mA
	Serial.println("8s radio.powerUp");
	delay(50);
	radio.powerUp();
	delay(8000);

// 28mA
	Serial.println("8s radio.powerDown");
	delay(50);
	radio.powerDown();
	delay(8000);

// 16.5mA
	Serial.println("8s radio.powerDown, LowPower.powerDown");
	delay(50);
	radio.powerDown();
	LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
	radio.powerUp();

// 16.5mA
	Serial.println("8s radio.stopListening, radio.powerDown, LowPower.powerDown");
	delay(50);
	radio.stopListening();
	radio.powerDown();
	LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
	radio.powerUp();

// 16.5mA
	Serial.println("8s radio.stopListening, LowPower.powerDown");
	delay(50);
	radio.stopListening();
	LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
}
````

TMRh20, 2016-04-13:
I think you have your answer. Those modules with glob of goo chips are not
great.

// https://github.com/rocketscream/Low-Power

gitgrimbo, 2016-04-13:
I was hoping that wouldn't be the answer!

Do you know if this is a widespread issue with the cheaper modules?  If so, can you recommend an alternative supplier?

Avamander, 2016-04-13:
Not cheaper modules. Clone modules are the ones causing issues. You might
get clones if you pay twice the much and you might get good originals if
you pay 1% more than you did. You never know until you carefully inspect
them and test them. The original ones usually perform a lot better with the
Transfer example (lower packet loss, higher speed) so just test, test and
again test.

gitgrimbo, 2016-04-13:
Thanks.  I've ordered previously from two eBay sellers, and one had a picture with the blob, and one had no blob, but both came with blobs!  The modules appear to work in terms of regular rx/tx behaviour, but now I'm trying a low-power project and noticed strange results.

So if I can't tell what I'm going to get from the seller's pictures do you have a seller that you use for reliable modules?

TMRh20, 2016-04-13:
Yeah, what @Avamander said. I can't recommend a supplier but I assume
mysensors.org might be a good place to look for RF24 hardware and hardware
advice. (Have not tried them myself)

gitgrimbo, 2016-04-13:
Thanks for your help guys.

Could there be a way of 'rescuing' my modules by turning them off instead of entering a low-power mode? (I've also just ordered a couple more of hopefully legit modules).

Avamander, 2016-04-13:
It's more hassle most likely than to get proper modules.

gitgrimbo, 2016-04-13:
@Avamander, sure I understand.

Out of interest, would either of you be able to run the sketch above and tell me what current readings you get.  If you don't have time, no problem, I'll just go by the values in [the chart you have pointed me at](https://github.com/TMRh20/RF24/issues/195#issuecomment-173091940).

dobrMAN, 2016-04-15:
Just for info. I have same issue with nRF24L01+ mini (SMD) [like this](url)
Anybody try this transmitters (SMD) in low power mode?

gitgrimbo, 2016-04-15:
@dobrMAN, is your link working? I get a 404.

dobrMAN, 2016-04-16:
@gitgrimbo may be [that](http://www.ebay.com/itm/5PCS-New-Mini-NRF24L01-SMD-1-27MM-RF-wireless-transceiver-module-/221917395219)?

dobrMAN, 2016-04-16:
And i found some different SMD [module](http://www.ebay.com/itm/New-Mini-NRF24L01-SMD-1-27MM-RF-wireless-transceiver-module-Small-Size-/162011278584) May be that can power down, but i cant try it now, delivery time around 2 month.

gitgrimbo, 2016-04-16:
I retried my test with new modules, http://www.amazon.co.uk/Professional-NRF24L01-Antenna-Wireless-Transceiver-5pcs/dp/B010N32SGM, and my low-power state now draws ~7.5mA.

7.5mA sounds about right ...

- 3.4mA (*5V@16Mhz Pro Mini in low-power state*) +
- 4mA (*nRF socket*) +
- 0.0009mA (*nRF powered down*) +
- innacuracies in measurement

So as @TMRh20 says, the eBay glob of goo chips seem to not perform to the nRF24L01+ datasheet specs.

I'm closing this issue as the problem is with my cheap eBay nRF24L01 modules and not the code.

closed 2016-04-16 by gitgrimbo