nRF24/RF24 #246: RF24 and Atxmega

JoLuk, 2016-04-18:
Hello,

Is it possible to use this library somehow with atxmega MCU? I will grateful for some sources if sutch exist about this matter. I tried to find anything but without success.

TMRh20, 2016-04-21:
Like https://github.com/TMRh20/RF24/pull/238 ? All Arduino supported boards should work as well.

akatran, 2016-04-21:
@JoLuk You can find the shared libraries projects for Atmel Studio 7 on the following comment
https://github.com/TMRh20/RF24/pull/238#issuecomment-205014554

akatran, 2016-04-22:
@JoLuk This libraries are working for ATXMEGA256D3 only. What MCU do you have?

JoLuk, 2016-04-22:
I using atxmega128a3u. I used this library when i writed sketches for arduino but now i change platforme for xmega and wondering how implement it for my projects. Thanks for comments i check all shared links and try do something with it.

akatran, 2016-04-22:
@JoLuk  Your MCU signature is \_\_AVR_ATxmega128A3U\_\_. 

So you can download this lib and change the line 45 
```#elif defined(__AVR_ATxmega256D3__)``` 
with 
```#elif defined(__AVR_ATxmega256D3__) || defined(__AVR_ATxmega128A3U__)``` 
in **RF24_config.h** and compile.

If the ports and hardware is the same it will play directly. If not the you have to check the difference between the two MCUs.

closed 2016-07-27 by TMRh20