nRF24/RF24 #248: getting started example extraneous "while"

gale3313, 2016-04-21:
In the pong-back section of TMRh20 "getting started" program, radio.read(&got_time, sizeof(unsigned long) is in a while (radio.available()) loop that is within an if (radio.available()) statement.   It appears that the while loop should be deleted.  See [http://forum.arduino.cc/index.php?topic=395481.new#new](url) for more info.

TMRh20, 2016-04-21:
Yeah, that is mostly intentional. Redundant code tends to catch the
attention of programmers. The main idea is to make users aware that there
is a limited radio buffer that needs to be emptied regularly or
'maintained' accordingly.

It also relates to whatever protocol you wish to implement:

1. Do you want the receiver to attempt a transmission for every payload
received or just the most recent?
2. Is the receiver running on Linux or Arduino? Note the delay(925);
in the linux
example <http://tmrh20.github.io/RF24/GettingStarted_8cpp-example.html> which
demonstrates/helps to demonstrate changes in behavior when the transmitter
is sending more often then 1payload/second.
3. On linux for example, using a while() loop prevents sync issues if both
devices are set to pause or check for payloads at set interval.
4. At high data rates, it is typically best to use a while() loop to clear
the buffer even if using interrupts, which prevents the receiver from
locking up due to missed interrupts.

Managing the RX buffer is very important.

gale3313, 2016-04-24:
Thank you for the reply.  I appreciate your explanation.  I haven’t found that sort of information in the various tutorials and websites at the top of a Google search on nrf24l01 and Arduino.

 

However, since it is redundant in the “getting started” example, then it seems to me it should be deleted, or else a comment should be added to explain why it is there.

 

No virus found in this message.
Checked by AVG - www.avg.com
Version: 2016.0.7539 / Virus Database: 4556/12072 - Release Date: 04/21/16

TMRh20, 2016-04-26:
I don't understand what the issue is? Is there some problem or confusion this is causing you?

Also, Whandall clearly does not understand From your thread:

> It is best to process all packets when they are available instead of just throwing them away,
reading them is the time consuming part of the processing

This is exactly what is happening, just that we only need the last received payload...

> I have never seen the receiver lock under overload.

Really? Try fill the RX buffer and see how many RX interrupts are received. This is not rocket science...

> I have never seen effects that point to a missing interrupt yet
and I run my variant of the driver interrupt based for a couple of months now.

This is very, very simple: anything that disables or requires the disabling of interrupts can cause missed interrupts.

I for one am always skeptical of programming 'experts' who spend all of their time posting to internet forums instead of writing code... (9.226 posts per day...)

closed 2016-04-26 by TMRh20

gale3313, 2016-04-27:
The issue is that the “getting started” example contains, using your words, “somewhat redundant” code that is not explained (commented).  

 

In addition, it contains inconsistent code (the while look is not used for all reads).

 

Both of those things can confuse newbies (I count myself in that group), and I think the “getting started” example is intended for newbies.

 

Hence my suggestion that you either eliminate the while loop or explain it.

 

 

 

From: TMRh20 [mailto:notifications@github.com] 
Sent: Monday, April 25, 2016 11:04 PM
To: TMRh20/RF24
Cc: gale3313; Author
Subject: Re: [TMRh20/RF24] getting started example extraneous "while" (#248)

 

I don't understand what the issue is? Is there some problem or confusion this is causing you?

TMRh20, 2016-04-27:
Well if you really think so, how about a real noob challenge?

You have my opinion and on why it is as it is. I challenge you to revise
and improve the example even more if preferred, with comments and/or code
and submit a pull request.

Noob up! I'm sure others will assist if you have questions.

Avamander, 2016-04-27:
@gale3313

If you create a pull request that makes the sketch really better, I'm fairly sure it'll get merged. It's also really simple to create a pull request.

Also, for the love of other Github users do not include the previous message when you use an email to comment here (because your client messes it up and Github can't autohide it). Also disable that AVG scan message, noone needs to know that.

galed3e3, 2016-04-28:
No way I'm taking that bait.

Sorry to reply with all the crap from previous messages/AVG.

TMRh20, 2016-04-28:
Bait?
You already 'put yourself out there' when you made your suggestions and the
'challenge' is not difficult.

Just put a single comment in a pull request, and if you want to take your
time and really look at improving it, that is just a further challenge to
make you not-a-noob and a contributor.

What is the worst I can do beyond saying no?

TalikJaafar, 2021-11-14:
Hi @TMRh20 
I know that this is an old discussion but I m a bit confused by the if/while combination used, I understand your opinion about the buffer and for me that make sens (using the while read() in the while will clean the buffers) and I wonder why you used the if condition , the while alone isn't enouth?

TMRh20, 2021-11-15:
I’m not sure what you are referring to. Could you post the code in question?