nRF24/RF24 #250: Failed to build Python wrapper on BeagleBone running Debian 8

hongquan, 2016-04-28:
I failed to build pyNRF24:

```
$ python3 setup.py build                              
running build
running build_ext
building 'RF24' extension
arm-linux-gnueabihf-gcc -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -march=armv7-a -mtune=cortex-a8 -mfpu=neon -mfloat-abi=hard -L/usr/local/lib -I/usr/local/include -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c pyRF24.cpp -o build/temp.linux-armv7l-3.4/pyRF24.o
cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
```

It failed with both Python 3 and Python 2, with the same error.

My system:

BeagleBone Black, running Debian 8.4

```
$ uname -a
Linux beaglebone 4.1.18-ti-r56 #1 SMP Thu Mar 31 00:22:03 UTC 2016 armv7l GNU/Linux
```

TMRh20, 2016-04-28:
Try waiting longer before killing the process.

hongquan, 2016-04-28:
I didn't kill. As you can see in the log, only the command `python3 setup.py build` is entered by me.
The processed was killed automatically by something when the build failed.

mz-fuzzy, 2016-04-28:
It seems that your bone is running out of resources, possibly ram. Your BBB has 512MB of ram, right? Try to increase swap, for reference see for example here https://github.com/lexruee/pi-switch-python/issues/13. Or alternativelly you can cross-compile (see #231).

hongquan, 2016-04-28:
Yes, I think so (RAM runs out).

closed 2016-04-28 by hongquan