nRF24/RF24 #252: Problem using the F() macro

nightswimmer, 2016-04-29:
I'm having some issues using this library when using the F() macro.
I has been using maniacbugs library for a few years, and now I updated to this one. The radio modules seem to work fine, but all the prints that use the F() macro come out as trash. If I remove the F() macros everything works out fine. This is an issue for me, since in the project I'm working on RAM is an issue. Tested the Getting started sketch plues a couple others with the same results.
Tried it in two new arduino installs (1.7.10 and 1.6.8) but the problem persists. :s Any ideas?

madleech, 2016-04-30:
Yes I've just run into this too when using platform.io. Seems there there is a clash in `RF24_config.h:132` when using PROGMEM.

To fix, change:
```
#elif defined(ARDUINO) && ! defined(__arm__) && !defined (__ARDUINO_X86__) && defined(XMEGA)
```

To:
```
#elif defined(ARDUINO) && ! defined(__arm__) && !defined (__ARDUINO_X86__) && !defined(XMEGA)
```

(or just remove the `defined(XMEGA)`  clause all together)

TMRh20, 2016-04-30:
This was due to a lack of proper testing on my part I guess but should be fixed now with the latest change. Thanks for identifying the problem!

akatran, 2016-04-30:
Should I introduce some compile time messages in order to see what is the compilation flow and what is not?

For example ```#pragma message "Using avr pgmspace"```.

Do you think it will flood the compilation log with too much info?

TMRh20, 2016-04-30:
Would be good for testing but I don't think we would want it printing
everything normally.

nightswimmer, 2016-04-30:
Great it's working now. Thank you for the quick fix. :)

akatran, 2016-04-30:
I can confirm that it doesn't break anything from my side.
The compiled output is exactly the same as before for XMega target.

I didn't have the time to test it on real hardware!

closed 2016-11-10 by TMRh20