nRF24/RF24 #255: No connection between RF24Gateway on RPi and Arduino after updating libraries

grahamcupr, 2016-05-11:
I've managed to get a decent connection between a base station (RPi) and a receiving station (Arduino), and it's been working very well for the last couple of weeks in my testing. 

However, I updated the libraries on the RPi and Arduino to the latest available versions, and now everything has stopped working. My setup is identical otherwise to what it was. I know this is not a hardware problem, as the radios work. I've done lots of other things to check the radios, arduino, Rpi. Any advice would be great, or anything else I could check. Please let me know if you require any further information. I really want to continue using this selection of libraries because they are fantastic.

TMRh20, 2016-05-12:
I guess if you changed only the software, then it should be a config issue of sorts, unless I broke something. 

I would first run the RF24 gettingstarted examples on the RPi and one node. See the examples using printf.h to get debug info. Get that working before anything else. (note: Disconnect pwr from modules to ensure fresh config due to conflicting settings)

Then see http://tmrh20.github.io/RF24Ethernet/ConfigAndSetup.html

You can enable debugging by editing the RF24Mesh_config.h files for example, to see more about what is going on.

grahamcupr, 2016-05-12:
Thanks for the response, I will take a look this evening and get back to you. Hopefully it will be ok.

grahamcupr, 2016-05-12:
I've got a little bit further forward. I've turned on the debugging in the header and I got this

start
27 MSH: Poll > -64dbm 
28 MSH: Got poll from level 0
30 MSH: Request address 
33 Set address:    2

so I can see that the address has been allocated on the mesh, and the ping trace from the Ping command on the raspi shows successful pings and as long as I keep the serial window open. As soon as I close the serial window, the pings stop. Which is a bit wierd because I have an ATMEGA chip on a custom circuit board that I programmed before I started having trouble, and that one I am able to ping successfully at any time. Sorry to be a pain!

TMRh20, 2016-05-12:
If using arduino IDE 1.6.8 update it. Else please test with an external pwr
supply or batteries.

grahamcupr, 2016-05-12:
Ah, I'm using arduino 1.6.8, I will update immediately.

grahamcupr, 2016-05-13:
My serial problem is solved, so thanks for that. Had no idea that Arduino 1.6.8 had that problem. Earlier in this thread I spoke about an ATMEGA based circuit board that still works with the RPI (which has not been updated since I started having trouble), in fact the connection has been rock solid. However with a second ATMEGA-based circuit board, to which I uploaded an updated RF24 sketch does not connect to the mesh at all. I have checked all the hardware I got the initial one working around February I think and works every time I use it. The chips I am using are ATMEGA328P-PU running at 1mhz. Is there a specific version of your libraries I need to use for this set up? I made the silly mistake of not noting what I was using in February when it was working. Sorry to keep pestering you, but I'm just really enthusiastic about these libraries and am desperate to get it working! :)

TMRh20, 2016-05-14:
>  Is there a specific version of your libraries I need to use for this set up?

Nope, unless a recent change broke something. You can check the commit list https://github.com/TMRh20/RF24/commits/master , browse files and download old commits, so you can always go back to old code to test.

grahamcupr, 2016-05-14:
I am going to have to admit defeat with this I'm afraid. I just can't replicate what I had before I updated the libraries. I've tried all your advice and I can't get anywhere. I've browsed through the list of commits, downgraded my Arduino to what I was using at the time, and I found the RF24 libraries I downloaded at the time, and nothing is working. I'll just have to go back to my fallback options. It's a shame because when the libraries worked, it was awesome. Never mind. Thanks for your help anyway,

Avamander, 2016-05-14:
Hardware change most likely. Do the examples themselves work? If not then
does printdetails print good results?

grahamcupr, 2016-05-15:
I believe I've narrowed down my problem. It's not a hardware change, as my hardware has been the same from the start of using the libraries. What I normally do is to build something on a breadboard with an Arduino and then trasport that onto a standalone ATMEGA328 chip, which is why I will have mentioned both these things in the flow of this conversation. Anyway, I had two problems by the looks of it, one which is now solved. Firstly the ATMEGA wasn't joining the mesh, but after running the Getting Started sketch and getting the ping/pong to work, I ran the RF24Mesh_Example sketch, it looks like the NRF module must have had some old config on or something, but anyway, I've now got it to successfully join the mesh, which leaves the final problem which I believe is with my modified sketch (the InteractiveServer_Mesh example). I think there is something wrong with this sketch and its not getting allocated an IP address. I'm a bit further forward now, so feel a bit happier. I'll let you know the outcome of my further testing when possible.

Avamander, 2016-05-15:
Yeah, it's a good idea to cycle the power if you encounter problems.

closed 2016-06-10 by Avamander