nRF24/RF24 #256: No instructions to install setuptools

gitgrimbo, 2016-05-12:
When following the instructions here, http://tmrh20.github.io/RF24/Python.html, on a Raspberry Pi B, I couldn't run this command

    ./setup.py build 

as I didn't have setuptools installed.

I had to use this command from here , https://pypi.python.org/pypi/setuptools#unix-wget

    wget https://bootstrap.pypa.io/ez_setup.py -O - | sudo python

mz-fuzzy, 2016-05-13:
good point, doc needs update for current configuration/installation for rpi/linuxes.

gitgrimbo, 2016-05-13:
I don't mind having a go at improving the docs myself (e.g. #254), but I'm still quite new to this project (and C/Python in general) so I have to be careful not to add duff information.

PaoloElMarinelo, 2016-10-13:
To install setuptools I follow this page :
https://packaging.python.org/installing/

And for me `.setup.py build` mean the setup.py I have to build is in this directory :`~/rf24libs/RF24/pyRF24`.
It's right ? however It works ...

closed 2020-03-29 by TMRh20