nRF24/RF24 #259: clearing EN_RXADDR

schmurfy, 2016-05-16:
I just got a weird issue, I was opening pipe 1 with an address but the data where arriving on pipe 2...
After making sure my code was correct which it appears to be I started digging and found out that the EN_RXADDR register was never cleare, only specific bits are set or reset.

The datasheet says that pipe 0 and 1 are enabled by default but in my case I had 0,1 and 2 enabled, is it my chip or is the normal behavior ? if it is it might be usefull to set it to 0 in begin.

TMRh20, 2016-05-18:
Typically it should only be an issue if you are using the default addresses of the chip. Otherwise, you would have had to have configured pipe 2 at some point with an address.Power cycle radios to clear settings, as they are retained. It helps saves space & memory for small devices.

schmurfy, 2016-05-18:
I don't remember enabling pipe 2 myself but I ran a lot of tests on these modules so it does not surprise me but what did suprise me a bit is that the previous configuration is saved when powered off.

What I ended up doing is:

```c
for(int i= 0; i< 6; i++){
  radio.closeReadingPipe(i);
}
```

I realize doing it in begin might not be a good idea if one want to keep the previous state as you said but It would just be nice to have a way to directly clear all the enabled bits without having to do 6 writes (maybe a `closeAllReadingPipes()`).
I can even submit a pull request if you are fine with it.

TMRh20, 2016-05-21:
I may have put in in an odd way, but the radios don't retain settings when powered off. If you look at the gettingstarted call response sketch, you will see the options for including printf.h and calling printf_begin. 

If you get a chance, please power cycle your radios, enable printf and use printDetails() to show the radio config & defaults for your device.

schmurfy, 2016-05-23:
I didn't understand either why settings were retained, I will check again.

Bitblade, 2016-09-25:
I experienced the same issue. 
I was experimenting with two Arduino Unos, one receiving, one transmitting. After switching the roles by reprogramming both Arduinos, I noticed the transmission speed was significantly lower (even though the data rate was still set to 1MBPS. 

While ruling out hardware issues, I noticed the problem disappeared when power was removed from the nRF24L01+. 

The output confirms not all registers are cleared by calling radio.begin()

Freshly power-cycled nRF24L01+
```
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xb11ee1d000 0xc2c2c2c2c2
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xb11ee1d000
RX_PW_P0-6       = 0x20 0x00 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x03
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
```

Switched from receiving to transmitting without power-cycle:
```
STATUS           = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1     = 0xb11ee1d000 0xb11ee1d000
RX_ADDR_P2-5     = 0xc3 0xc4 0xc5 0xc6
TX_ADDR          = 0xb11ee1d000
RX_PW_P0-6       = 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA            = 0x3f
EN_RXADDR        = 0x02
RF_CH            = 0x4c
RF_SETUP         = 0x07
CONFIG           = 0x0e
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01+
CRC Length       = 16 bits
PA Power         = PA_MAX
```

It shouldn't be an issue that `RX_ADDR_P0-1` is not reset, but I did assume `EN_RXADDR` & `RX_PW` (_P0-6) would be reset by a call to begin()

TMRh20, 2016-11-10:
This sounds like normal behavior. 
The begin() function does not attempt to clear all settings on the chip, only some main ones. 
Only removing power will clear all settings.

closed 2016-11-10 by TMRh20