nRF24/RF24 #260: RF24 + FPGA = WIFI?

danxn, 2016-05-27:
Hello, TMRh20!

Sorry for a bit offtop, but my question is close to what you do, so may be you can help me. Do you know - is it possible to use RF24 chip in connection with FPGA module to implement own WiFi chipset? As i know RF24 implements all RF stuff, so FPGA can implement rest of WiFi - it's logic level. With cheap FPGA's like Cyclone or Spartan it looks intresting for IoT applications. What do you think?

Avamander, 2016-05-28:
Shortly put, no, never. You would have to modify the dye itself. nRF24
definitely lacks in both hardware features and control available to you to
emulate something WiFi like.

danxn, 2016-05-28:
Thanks :) Ok, it's not a problem, i think. These are a lot of other RF chips :) This chip just better known than others.

danxn, 2016-05-28:
@Avamander Excuse me, may I ask you - does FPGA able to implement in hardware TCP/IP RF layer different from WiFi better than it can be implemented on Atmel microcontroller?

closed 2016-06-10 by Avamander