nRF24/RF24 #261: RF24 and DHT11 together run problem.

emrekilic, 2016-05-28:
I am using rf24 library without any problem with rpi 3. Also i am using DHT11 thermometer and its working. I can read temperature data. But if i run rf24 codes first, then i cant get temperature data from DHT11 module. I didnt understand why rf24 effects DHT module. Dht module using gpio4 for data. And also needs bcm2835 library. Anyone can explain how to avoid rf24 library to effect DHT11 library.

akatran, 2016-05-28:
I also have the same setup with RPi 2 without problem using DHT11 and rf24 at the same time.
Both programs run as service at system boot with out conflicts.

It shouldn't make a conflict, after all RF24 uses spi and and DHT is using just a pin as you mentioned. 

If you start the program for rf24 and stop it, can you then read data from DHT module;

emrekilic, 2016-05-28:
yes i can read it, if i stop program for rf24

closed 2016-06-10 by Avamander

mcodo, 2019-08-20:
I know this is an old thread, but did you find any solution for this?
I'm having the same problem when using mesh.write.