nRF24/RF24 #262: Odroid C1 Support

ingamedeo, 2016-05-28:
Hi,
I was wondering whether it would be possible to support the **Odroid C1** dev board.

SPI works with SPIDEV (/dev/spidev0.0) but I am not able to talk to the RF module.

```
sudo ./spidev_test -D /dev/spidev0.0
spi mode: 0
bits per word: 8
max speed: 500000 Hz (500 KHz)

FF FF FF FF FF FF 
40 00 00 00 00 95 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
FF FF FF FF FF FF 
DE AD BE EF BA AD 
F0 0D
```

When running the **getting started** example I get:

```
sudo ./gettingstarted
STATUS           = 0x00 RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=0
RX_ADDR_P0-1     = 0x0000000000 0x0000000000
RX_ADDR_P2-5     = 0x00 0x00 0x00 0x00
TX_ADDR          = 0x0000000000
RX_PW_P0-6       = 0x00 0x00 0x00 0x00 0x00 0x00
EN_AA            = 0x00
EN_RXADDR        = 0x00
RF_CH            = 0x00
RF_SETUP         = 0x00
CONFIG           = 0x00
DYNPD/FEATURE    = 0x00 0x00
Data Rate        = 1MBPS
Model            = nRF24L01
CRC Length       = Disabled
PA Power         = PA_MIN
```

**nrf24l01+ - Odroid wiring** is as follows:

3.3V > 3.3V (1)
GND > GND (6)
MISO > MISO (#106)
MOSI > MOSI (#107)
SCK > SCLK (#105)
CSN > CE0 (#117)
CE > GPIO (#115)

Odroid C1 GPIO Header is  [here](http://cdn.overclock.net/b/b6/b6755fac_ss2014-12-27at05.02.37.png
)

Thanks for your help,
Cheers from Italy,
Amedeo

Avamander, 2016-05-28:
Are you sure the MISO and MOSI don't have to be reversed?

ingamedeo, 2016-05-28:
@Avamander No, MOSI is Master Out / Slave In and MISO Master In / Slave Out, so they shouldn't be reversed.

Avamander, 2016-05-28:
Yeah, but MOSI Master out (slave in) has to be connected to MISO Master in
(slave out) ;) It now depends on how they are labelled.

ingamedeo, 2016-05-28:
I've tried already and it still doesn't work. Same result.

Avamander, 2016-05-28:
Sorry my bad, I was thinking about a device that has the lines reversed.

ingamedeo, 2016-05-30:
The wiring is correct. I've checked many times today.

razerraz, 2016-10-07:
It's working for me using RF24(103,0) with python lib, RF24 CE pin to 22(103) and CS to 24(CE0/117)
Be sure to load spicc kernel module

closed 2016-11-10 by TMRh20