nRF24/RF24 #264: Atmega328p @ 8mhz @ 250kbps = fail

cimba007, 2016-06-05:
int pipe = wait(1500);
    if(pipe != -1)
    {
      uint8_t tc = radio.testRPD();

Using   Serial << "setDataRate: " << radio.setDataRate(RF24_250KBPS) << endl;

tc is always 0 and I get a very very poor reception. Using the same code with RF24_2MBPS my results are fine. 1MBPS is working great too. Have I missed an important point?

My whole init:
`
void rf_setup()
{
 radio.begin();
  radio.enableDynamicPayloads();
  radio.setAutoAck(1);                    // Ensure autoACK is enabled
  //radio.setChannel(100);
  //radio.setPALevel(RF24_PA_MAX);
  Serial << "setDataRate: " << radio.setDataRate(RF24_2MBPS) << endl;
  radio.setCRCLength(RF24_CRC_16);
  radio.setRetries(15,15); // 5 retries with 15 * 250µS = 3.75ms delay between each

  if(settings[ce::myid] == 0){
    Serial.println(F("I am station"));
    // Station
    radio.openWritingPipe(station_to_node[0]);
    radio.openReadingPipe(1,node_to_station[1]);         
    radio.openReadingPipe(2,node_to_station[2]);      
    radio.openReadingPipe(3,node_to_station[3]);      
    radio.openReadingPipe(4,node_to_station[4]);      
    radio.startListening();                           
  }else{
    Serial.print(F("I am node "));
    Serial.println(settings[ce::myid]);
    // Node
    radio.openWritingPipe(node_to_station[settings[ce::myid]]);
    radio.openReadingPipe(1,station_to_node[settings[ce::myid]]);
  }
}
`

DasHammer, 2016-06-05:
I am having the same exact problem with the atmega 328p.

TMRh20, 2016-06-05:
How are you powering the radios? 250kbps on-air time is 8 times longer than
2mbps, so uses more power.

<https://github.com/TMRh20/RF24/issues/264#issuecomment-223829181>, or mute
the thread

cimba007, 2016-06-05:
I use an 250mA LDO with 3300µF cap on my receiving node and 1500µF on my sending node. I will hook up my osci to check the voltage.

thx TMRh20, will reply soon

cimba007, 2016-06-05:
okay .. I measured Peak-to-Peak ripple of 60mV during send. Adding another 1500µF on my sending node (3000µF parallel) decreases the ripple to 30mV (only when sending).

Still the same Problem. I replaced the PA+LNA-Board with a "normal" one and got still the same figures.

Avamander, 2016-06-05:
Okay, now, if you take the example sketch and only add the line to lower
tha datarate, does it work?

cimba007, 2016-06-05:
I will try this out tomorrow. thanks for your help

cimba007, 2016-06-06:
Nothing changed .. contrary to what might be sane I get the worst results on 250kbps.

I changed the example sketch to include RPD:

http://pastebin.com/2V1hvDQY

I looked many hours into the source and found nothing .. My last resort would be to completly redesign my pcb although oscilloscope reading show nothing wrong with the power supply.

I even powered both nodes from an 3,7V -> LDO -> nrf24 .. giving even more stable supply but simply .. no ..

TMRh20, 2016-06-06:
A couple of questions:

1. Since the title is 'Atmega328p @ 8mhz' have you tested with any 16mhz 328s? Difference or no difference?
2. How about if you comment out [lines 111, 112 & 113](https://github.com/TMRh20/RF24/blob/master/examples/GettingStarted/GettingStarted.ino#L111) in the gettingstarted example and just do one-way communication? Please post the serial output from both devices.

cimba007, 2016-06-08:
I narrowd down the problem to the auto-ack feature although still under investigation. I use these settings:

  radio.setAutoAck(true);
  radio.setRetries(15,15);

radio.setAutoAck(false); ==> seems to work better

remember .. 1mbps and 2mbps works fine ..somehow autoack doesn't work as expected on 250kbps.

I will test your two suggestions now.

cimba007, 2016-06-08:
mycode: http://pastebin.com/V9t1MYVR
changing baud-rate for different tests

Test 1: NO OK
- Both modules @ 8mhz internal oscillator
- Both modules set to 250kbps
- Reply code removed so the communication is just "node1 (ping) -> node2 (pong)" 
- Node1 and Node2 are PA+LNA with a distance of 15-20cm, stable supply and RF24_PA_MAX

Output on node2: http://pastebin.com/ATssyu2Y

Test 2: OK
- all the same as Test1, both modules set to 1mbps

Output on node2: http://pastebin.com/W1p0mqZ8

Test 3: OK
- same as Test1, "radio.setAutoAck(false);"

Output on node2: http://pastebin.com/JD8DLw4Z

Test 4: NOT OK
- same as Test1 but "radio.setRetries(15,15);"

Output on node2: http://pastebin.com/zPkGbJMv

Further tests are to switch back to 16mhz crystal

trojanc, 2016-06-08:
Hi

I'm not that technical about the inner workings...and do not have the solution for your problem,  but I've had endless issues with RF24_PA_MAX when the modules are too close to each other (like in the same room). Somehow they interfere with each other. When the modules are further away I use RF24_PA_MAX, otherwise RF24_PA_MIN while developing with the modules close by....don't know if that help :smiley:

cimba007, 2016-06-08:
hi trojanc, thanks for this tip. I don't want to retry all tests but I did for Test1 and moved the sending node 1 room away, all settings staying the same.

Avamander, 2016-06-10:
This again sounds like you have stumbled upon counterfeit modules that perform not as they are supposed to. Clones tend to have issues with autoack.

closed 2016-06-13 by Avamander