nRF24/RF24 #266: Throughput Question

trademark18, 2016-06-22:
I am simply curious about what the bottleneck is for this implementation's throughput.  The datasheet for the nRF24L01 is something up to 2Mbps, and we see that setting in the code, yet at best I get 35kbps.  I'm aware that the datasheet rating is not what should be expected in reality, but this is a significant difference, and I'm curious to see where improvements might be made.  

Other libraries seem to have similar throughput numbers (from what I've seen) so I assume there is a good reason for it.  If someone knows what that reason is, I'd be happy to hear it.  Thanks!

Here's what I learned by asking on the Arduino Forum
https://forum.arduino.cc/index.php?topic=407017.0

Thanks for the hard work on this library!

TMRh20, 2016-06-22:
Easy question: Auto-ack halves the transfer rate:

https://www.youtube.com/watch?v=mODflsK1Ab0

http://tmrh20.github.io/RF24/Transfer_8ino-example.html

trademark18, 2016-06-22:
Excellent video.  I didn't realize that the ACK packets were quite so hard on the overall throughput.  Thanks!

closed 2016-06-22 by trademark18