nRF24/RF24 #268: Intel Edison fails compile

acidhax, 2016-07-01:
I've added #define MRAA and #define RF24_LINUX to the RF24_Config.h user defined section.

`root@eddybeddy:~/RF24# ./configure
[SECTION] Detecting arm compilation environment.
  [INFO] arm-linux-gnueabihf-gcc not found.
  [INFO] arm-linux-gnueabihf-g++ not found.
[SECTION] Detecting target machine.
[OK] machine detected: SoC=unknown, Type=unknown, CPU=i686.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:SPIDEV.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[SECTION] Cleaning previous builds.
[OK] Finished.
root@eddybeddy:~/RF24# make
g++ -fPIC -Ofast -Wall -pthread  -c RF24.cpp
RF24.cpp: In member function 'bool RF24::begin()':
RF24.cpp:580:9: error: 'class GPIO' has no member named 'begin'
    gpio.begin(ce_pin,csn_pin);
         ^
Makefile:40: recipe for target 'RF24.o' failed
make: *** [RF24.o] Error 1`

mz-fuzzy, 2016-07-01:
undo what you've done to that poor file, remove the cat from your head and see ./configure --help :-)

```
./configure --driver=MRAA
make
```

acidhax, 2016-07-01:
Wellp, that was easy. Works like a charm.

closed 2016-07-01 by acidhax