nRF24/RF24 #272: A public method to check if the chip is connected

wmarkow, 2016-07-19:
Would it be a good enhancement to expose a public method that will check if the nRF24 chip is connected to the SPI bus and powered on?
As discussed in #213, when the chip is not connected (or is damaged or so) then SPI reads mostly 0x00 or 0xFF value from it. To check if the chip is really connected and running we could read the value of some specific register. By some specific register I mean the register that we know for sure that its value MUST be different than 0x00 or 0xFF. We could use a **SETUP_AW** register as according to the specs its value will never be 0x00 nor 0xFF:
```
SETUP_AW = 0b000000xx,
where
xx is 01 or 11 or 10 
xx = 00 is illegal
```

I would appreciate that method (like **bool RF24::isChipConnected()**) as it will make it easier to check for chip's hot insertions/removals without calling the **begin()** method.

TMRh20, 2016-07-24:
I don't see much harm in something like this, except would it be useful to include it in the [isValid()](http://tmrh20.github.io/RF24/classRF24.html#a35e5f1533b7753806c42b76e782d917e)  function?

Krastanov, 2016-12-04:
I have started using this patch for my work. It would be great if it could be accepted in the official version.

MDM63, 2016-12-06:
This would have saved me great deal of time and work. Would be good addition in the library.

wmarkow, 2017-05-04:
Pull request has been accepted and merged.

closed 2017-05-04 by wmarkow

ortegafernando, 2017-10-08:
Hi, when could you use this function? I mean, I always configure all the stuff arround RF24 in my SETUP. But I don't know if you could check in the LOOP part if "isChipConnected" and if it is not, reset it (may be mosfet) and reconfigure it again.

An example will be great, thanks.

soligen2010, 2017-10-09:
I use it in setup only to see if the chip is connected.  Chips wouldn't normally get connected/disconnected while the program runs.

For example, I have the option to have one or two connected for optional diversity.  In setup I check for both, then use whichever one is present, or use both for receiver diversity if they both exist.  This way I don't have to edit code for the various configurations I support.

ortegafernando, 2017-10-09:
> Chips wouldn't normally get connected/disconnected while the program runs.

Sometimes I get my module "blocked" so I thought that with this new function I could check if it is "death" or "alive". I think my problem has another source, hasn't it ?

soligen2010, 2017-10-10:
Yes I think your issue is something else. If the chip is "crashing" then
maybe poor power supply or perhaps for some reason the tx and or rx
question are filling. Maybe something else but these are the first things
that come to mind

<https://github.com/nRF24/RF24/issues/272#issuecomment-335261821>, or mute
the thread