nRF24/RF24 #274: wiringpi and RF24 in the same module

MikeMitterer, 2016-07-22:
I'm missing something or is it not possible?

```cpp
...
#include <RF24/RF24.h>
#include <wiringPi.h>

...
int main(void) {
...
// INPUT is not defined if compiling for Raspi
pinMode(gpioButton, INPUT);

// this is what I want:
        if (digitalRead(gpioButton) == 0) {
            delay(50);
            if (digitalRead(gpioButton) == 0) {
                printf("Button pressed!\n");
                counter++;
            }
        }
}
```
Including wiringPi.h leads to all kind of error messages like:
```bash
In file included from /Volumes/Daten/DevLocal/DevCPP/Projects/RaspiHelloWorld/main.cpp:5:0:
/Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/include/wiringPi.h:227:44: error: macro "millis" passed 1 arguments, but takes just 0
 extern unsigned int millis            (void) ;
                                            ^
In file included from /Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/utility/includes.h:7:0,
                 from /Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/RF24_config.h:38,
                 from /Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/RF24.h:18,
                 from /Volumes/Daten/DevLocal/DevCPP/Projects/RaspiHelloWorld/_shared/rf24/rf24.hpp:13,
                 from /Volumes/Daten/DevLocal/DevCPP/Projects/RaspiHelloWorld/main.cpp:4:
/Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/utility/RPi/RF24_arch_config.h:40:57: error: conflicting declaration of C function ‘void bcm2835_gpio_fsel(int, int)’
   #define pinMode(pin,value) bcm2835_gpio_fsel(pin,value);
                                                         ^
In file included from /Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/utility/includes.h:6:0,
                 from /Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/RF24_config.h:38,
                 from /Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/RF24.h:18,
                 from /Volumes/Daten/DevLocal/DevCPP/Projects/RaspiHelloWorld/_shared/rf24/rf24.hpp:13,
                 from /Volumes/Daten/DevLocal/DevCPP/Projects/RaspiHelloWorld/main.cpp:4:
/Volumes/Daten/DevLocal/DevCPP/raspberry/rootfs/usr/local/include/RF24/utility/RPi/bcm2835.h:1172:17: note: previous declaration ‘void bcm2835_gpio_fsel(uint8_t, uint8_t)’
     extern void bcm2835_gpio_fsel(uint8_t pin, uint8_t mode);
```

What I want is to read! from GPIO.
digitalWrite works with your lib but `digitalRead` and `"INPUT"` is not defined.
Thats why I tried it with wiringPi...

MikeMitterer, 2016-07-22:
Found a solution:
```cpp
btnState = bcm2835_gpio_lev((uint8_t) gpioButton);
```
Is there a reason for not declaring something like:
```cpp
uint8_t digitalRead(const uint8_t pin) { return bcm2835_gpio_lev(pin); }
```
What I figured out is that if I set `pinMode(gpioButton, BCM2835_GPIO_FSEL_INPT);` it crashes the whole program.

MikeMitterer, 2016-07-22:
This solved my problem:
```cpp
main() {
    if(!bcm2835_init()) {
        printf("GPIO initialization failed!\n");
        return 1;
    }
    printf("RASPI is up and running\n");
   ...
}
```
OK - it was me but I find it quite confusion that some "Arduino-functions" are implemented but others like digitalRead is missing...

TMRh20, 2016-07-24:
Yes, it is probably confusing, since nobody has made or requested a wiringPi specific 'port' of RF24, it only includes the required functions for the code + examples to be compatible.

closed 2016-11-10 by TMRh20

yp25, 2016-11-30:
I still have a problem with #include<wiringPi.h> when working with #include<RF24/RF24.h>
before adding include wiringPi , my code was fine, and I able to play with nrF24L01 module with raspberry Pi2. but after I just add #include <wiringPi.h> there was many error like 
`macro "milis" passed 1 arguments, but takes just 0`
`declaration of static void GPIO::open(int,int) outside of class is not definition [-permissive]`

My aim is add button as input and my code will read button state and then send through RF24. Any idea how to solve this ?

yp25, 2016-11-30:
sorry, I think I understand now about solution with bcm2835 library which @MikeMitterer written above. Thank you

mchukurs, 2019-12-06:
> 
> 
> sorry, I think I understand now about solution with bcm2835 library which @MikeMitterer written above. Thank you

May I ask you what you changed?