nRF24/RF24 #275: error: 'PORT_t' does not name a type

MikeMitterer, 2016-07-29:
Updated with platformio to 973533d799 and now I get the above error...

MikeMitterer, 2016-07-29:
Sorry - forgot to post my PIO-Settings:

[env:pro16MHzatmega328]
platform = atmelavr
framework = arduino
board = pro16MHzatmega328

After downgrading to the prev version, the one with arch-folder instead of utilities-folder, it finds it definitions again.

Avamander, 2016-07-29:
To find out what the issue is you need to post more than that line of the error message.

jrobeson, 2016-08-12:
I can't be sure it's the same cause as they see, but this is with https://github.com/TMRh20/RF24/commit/973533d799a1973bc558f885093a0259596f10a5 as well

```sh
 ~/w/l/s/arduino_target  platformio run                                                                   2057ms  2016-08-11 23:43:18
[Thu Aug 11 23:43:37 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
avr-g++ -o .pioenvs/uno/src/main.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility -Isrc src/main.cpp
avr-g++ -o .pioenvs/uno/RF24_ID433/RF24.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/RF24.cpp
avr-gcc -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c
avr-g++ -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp
avr-gcc -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c
In file included from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.h:20:0,
from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:9:
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.h:83:1: error: 'PORT_t' does not name a type
PORT_t * GPIO_getPort(int pinnum, uint8_t * pin_bm);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In static member function 'static void GPIO::open(int, int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In function 'int read(int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In static member function 'static void GPIO::write(int, int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In function 'int read(int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:35:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
scons: *** [.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.o] Error 1
In file included from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c:8:0:
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.h:83:1: error: unknown type name 'PORT_t'
PORT_t * GPIO_getPort(int pinnum, uint8_t * pin_bm);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c:13:1: error: unknown type name 'PORT_t'
PORT_t * GPIO_getPort(int pinnum, uint8_t * pin_bm)
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c: In function 'GPIO_getPort':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c:16:2: error: unknown type name 'PORT_t'
[... cut ...]
PMIC.CTRL = PMIC_LOLVLEN_bm | PMIC_MEDLVLEN_bm | PMIC_HILVLEN_bm;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:50:51: error: 'PMIC_HILVLEN_bm' undeclared (first use in this function)
PMIC.CTRL = PMIC_LOLVLEN_bm | PMIC_MEDLVLEN_bm | PMIC_HILVLEN_bm;
^
scons: *** [.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.o] Error 1
scons: *** [.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.o] Error 1
===================================================== [ ERROR ] Took 0.76 seconds ====================================================
```

Avamander, 2016-08-12:
What boards are you using? Maybe you need to fix the board file to actually use proper includes.

jrobeson, 2016-08-12:
It's an uno.  It compiled successfully with a previously version published on platformio.

Avamander, 2016-08-12:
I'll release a new version, update, if the issue persist let me know.

jrobeson, 2016-08-12:
I'm not too familiar with how platformio handles versioning. Do you have to tell it about the new version?

```sh
$ platformio update
Platform atmelavr
--------
Updating [ 433 ] RF24 library:
Versions: Current=973533d799, Latest=973533d799 	 [Up-to-date]
```

Avamander, 2016-08-12:
It will take a few hours I suspect.

jrobeson, 2016-08-13:
same issue. a duo isn't an ATXMega256D3 (right?), so should these files even be included?

jrobeson, 2016-08-13:
ah, i missed the code in `RF24_config.h`.

jrobeson, 2016-08-13:
I think I only saw the same issue to to cached dependencies or something. I cleared that out. Now there's a new, but similar one

```sh
platformio run
[Fri Aug 12 22:50:35 2016] Processing uno (platform: atmelavr, board: uno, framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------
avr-gcc -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c
avr-g++ -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.o -c -fno-exceptions -fno-threadsafe-statics -std=gnu++11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c: In function '__start_timer':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:37:2: error: 'TCE0' undeclared (first use in this function)
TCE0.CTRLA = TC_CLKSEL_DIV256_gc;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:37:2: note: each undeclared identifier is reported only once for each function it appears in
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:37:15: error: 'TC_CLKSEL_DIV256_gc' undeclared (first use in this function)
TCE0.CTRLA = TC_CLKSEL_DIV256_gc;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:40:15: error: 'TC_WGMODE_NORMAL_gc' undeclared (first use in this function)
TCE0.CTRLB = TC_WGMODE_NORMAL_gc;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:47:18: error: 'TC_OVFINTLVL_HI_gc' undeclared (first use in this function)
TCE0.INTCTRLA = TC_OVFINTLVL_HI_gc;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:50:2: error: 'PMIC' undeclared (first use in this function)
PMIC.CTRL = PMIC_LOLVLEN_bm | PMIC_MEDLVLEN_bm | PMIC_HILVLEN_bm;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:50:14: error: 'PMIC_LOLVLEN_bm' undeclared (first use in this function)
PMIC.CTRL = PMIC_LOLVLEN_bm | PMIC_MEDLVLEN_bm | PMIC_HILVLEN_bm;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:50:32: error: 'PMIC_MEDLVLEN_bm' undeclared (first use in this function)
PMIC.CTRL = PMIC_LOLVLEN_bm | PMIC_MEDLVLEN_bm | PMIC_HILVLEN_bm;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/compatibility.c:50:51: error: 'PMIC_HILVLEN_bm' undeclared (first use in this function)
PMIC.CTRL = PMIC_LOLVLEN_bm | PMIC_MEDLVLEN_bm | PMIC_HILVLEN_bm;
^
avr-gcc -o .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.o -c -std=gnu11 -g -Os -Wall -ffunction-sections -fdata-sections -mmcu=atmega328p -DF_CPU=16000000L -DPLATFORMIO=021102 -DARDUINO_ARCH_AVR -DARDUINO_AVR_UNO -DARDUINO=10608 -I.pioenvs/uno/FrameworkArduino -I.pioenvs/uno/FrameworkArduinoVariant -I.pioenvs/uno/SPI -I.pioenvs/uno/RF24_ID433 -I.pioenvs/uno/RF24_ID433/utility .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.c
In file included from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.h:20:0,
from .pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:9:
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio_helper.h:83:1: error: 'PORT_t' does not name a type
PORT_t * GPIO_getPort(int pinnum, uint8_t * pin_bm);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In static member function 'static void GPIO::open(int, int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:15:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In function 'int read(int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:33:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp: In static member function 'static void GPIO::write(int, int)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:2: error: 'PORT_t' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:11: error: 'p' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/gpio.cpp:40:37: error: 'GPIO_getPort' was not declared in this scope
PORT_t * p = GPIO_getPort(port,&pin);
^
[... 85 lines cut ...]
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/spi.cpp:41:2: error: 'port' was not declared in this scope
port->DIRCLR = SPI_MISO_bm;
^
[... cut ...]
device->CTRL = SPI_ENABLE_bm | SPI_MASTER_bm | SPI_MODE_0_gc | SPI_PRESCALER_DIV4_gc;
^
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/spi.cpp: In member function 'uint8_t SPI::transfer(uint8_t)':
.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/spi.cpp:37:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
scons: *** [.pioenvs/uno/RF24_ID433/utility/ATXMegaD3/spi.o] Error 1
===================================================== [ ERROR ] Took 0.48 seconds =====================================================
```

Avamander, 2016-08-13:
Somehow, your compiler is including files it should not. That happens when one of these is defined 
```
#if defined(__AVR_ATxmega64D3__) || defined(__AVR_ATxmega128D3__) || defined(__AVR_ATxmega192D3__) || defined(__AVR_ATxmega256D3__) || defined(__AVR_ATxmega384D3__) // In order to be available both in windows and linux this should take presence here.
…	
37	  #include "utility/ATXMegaD3/RF24_arch_config.h"
```

This is an issue with platformio or your install, not RF24.

jrobeson, 2016-08-14:
I filed a bug with platformio linking to this bug.  You can probably close it now.

closed 2016-08-15 by Avamander