nRF24/RF24 #276: Compatibility and wiring with Digispark (Not Digispark Pro)

bilogic, 2016-08-11:
Hi,

With reference to #226, can I use this library to connect the nRF24L01 and an adapter with a Digispark? If yes, how should they be wired together? Thank you

(The links below are to show exactly what I have, let me know to remove them if I'm breaking any rules.)

Digispark: http://www.ebay.com/itm/5PCS-Mini-ATTINY85-Micro-USB-Development-Board-for-Digispark-Kickstarter-/221926062050?hash=item33abd2f7e2:g:LrEAAOSw14xWMY0O
Adapter: http://www.ebay.com/itm/Socket-Adapter-plate-Board-for-8Pin-NRF24L01-Wireless-Transceive-module-51-/200960749614
nRF24L01: http://www.ebay.com/itm/5pcs-NRF24L01-2-4GHz-Antenna-Wireless-Transceiver-Module-/201414965820?hash=item2ee5447a3c:g:gUQAAOSwLa9UXJmy

Avamander, 2016-08-15:
The Digispark board uses an ATtiny85, that is supported by this library. You will have basically no free pins. Assuming you are using a good install of necessary board files, you should not have to add  `#define __AVR_ATtiny85__` if not, then try adding that line to the beginning of your sketch.

 https://arduino-for-beginners.blogspot.com/2013/04/process-of-getting-attiny85-to-work.html

bilogic, 2016-08-15:
@Avamander thanks for the info and links. The lack of free pins not a problem as my objective is a USB HID receiver. I have already managed to get it connected as a USB HID (keyboard).

Since some of the ATtiny85 pins on the Digispark are used to connect to the USB host, I'm confused about how to wire a Digispark to nRF24L01. This Mirf and SPI85 is a little beyond my limits of understanding.

I think my question is, is it possible to have the Digispark functioning concurrently as a USB HID and receive on the nRF24L01 at the same time? (I understand Digispark and USB HID is not the scope of this library, but I could not find a better place to ask, so thank you for any help)

Avamander, 2016-08-15:
https://tmrh20.github.io/RF24/ATTiny.html shows how exactly to connect the NRF24L01+ to a ATtiny85, if you need any pins left free, you have to build circuit described at both the previous link and this one https://nerdralph.blogspot.com.ee/2014/01/nrf24l01-control-with-3-attiny85-pins.html.

closed 2016-11-10 by TMRh20

XaviTorello, 2017-11-12:
@bilogic, please could you provide some feedback about the integration of this with a Digispark?

Many thanks

XaviTorello, 2017-11-15:
Ok, thanks @bilogic