nRF24/RF24 #278: printDetails() not working on nano clone.

patchido, 2016-08-14:
hi,

im trying to debug an application and for some reason the printdetails wont work, i know that this issue has been brought up some times now but i don't know what else to do, this is why i am trying

```
Serial.begin(115200);
  Serial.println(F("test"));
  printf_begin();
  Serial.println(F("test2"));
  radio.printDetails();
  Serial.println(F("test3"));
```

my serial is returning test and test2, but never gets to test3 meaning that in printdetails is somewhat looping or stuck.

the atmega im using is, [iteaduino nano](https://www.itead.cc/iteaduino-nano.html)

TMRh20, 2016-08-14:
Did you #include <printf.h> ?

Alicetech, 2016-08-18:
printDetails() does not work at all on Nano. Below is an example

`
/*
* Getting Started example sketch for nRF24L01+ radios
* This is a very basic example of how to send data from one node to another
* Updated: Dec 2014 by TMRh20
*/

#include <printf.h>
#include <SPI.h>
#include "RF24.h"

/****************** User Config ***************************/
/***      Set this radio as radio number 0 or 1         ***/
bool radioNumber = 0;

/* Hardware configuration: Set up nRF24L01 radio on SPI bus plus pins 7 & 8 */
RF24 radio(9,10);
/**********************************************************/

byte addresses[][6] = {"1Node","2Node"};

// Used to control whether this node is sending or receiving
bool role = 0;

void setup() {
  Serial.begin(115200);
  Serial.println(F("RF24/examples/GettingStarted"));
  Serial.println(F("*** PRESS 'T' to begin transmitting to the other node"));
  
  radio.begin();

  // Set the PA Level low to prevent power supply related issues since this is a
 // getting_started sketch, and the likelihood of close proximity of the devices. RF24_PA_MAX is default.
  radio.setPALevel(RF24_PA_LOW);
  
  // Open a writing and reading pipe on each radio, with opposite addresses
  if(radioNumber){
    radio.openWritingPipe(addresses[1]);
    radio.openReadingPipe(1,addresses[0]);
  }else{
    radio.openWritingPipe(addresses[0]);
    radio.openReadingPipe(1,addresses[1]);
  }
  
  // Start the radio listening for data
  radio.startListening();
  
    Serial.println(F("printDetails"));
  radio.printDetails();
    Serial.println(F("printDetails End"));
}

void loop() {
  
  
/****************** Ping Out Role ***************************/  
if (role == 1)  {
    
    radio.stopListening();                                    // First, stop listening so we can talk.
    
    
    Serial.println(F("Now sending"));

    unsigned long start_time = micros();                             // Take the time, and send it.  This will block until complete
     if (!radio.write( &start_time, sizeof(unsigned long) )){
       Serial.println(F("failed"));
     }
        
    radio.startListening();                                    // Now, continue listening
    
    unsigned long started_waiting_at = micros();               // Set up a timeout period, get the current microseconds
    boolean timeout = false;                                   // Set up a variable to indicate if a response was received or not
    
    while ( ! radio.available() ){                             // While nothing is received
      if (micros() - started_waiting_at > 200000 ){            // If waited longer than 200ms, indicate timeout and exit while loop
          timeout = true;
          break;
      }      
    }
        
    if ( timeout ){                                             // Describe the results
        Serial.println(F("Failed, response timed out."));
    }else{
        unsigned long got_time;                                 // Grab the response, compare, and send to debugging spew
        radio.read( &got_time, sizeof(unsigned long) );
        unsigned long end_time = micros();
        
        // Spew it
        Serial.print(F("Sent "));
        Serial.print(start_time);
        Serial.print(F(", Got response "));
        Serial.print(got_time);
        Serial.print(F(", Round-trip delay "));
        Serial.print(end_time-start_time);
        Serial.println(F(" microseconds"));
    }

    // Try again 1s later
    delay(1000);
  }

/****************** Pong Back Role ***************************/

  if ( role == 0 )
  {
    unsigned long got_time;
    
    if( radio.available()){
                                                                    // Variable for the received timestamp
      while (radio.available()) {                                   // While there is data ready
        radio.read( &got_time, sizeof(unsigned long) );             // Get the payload
      }
     
      radio.stopListening();                                        // First, stop listening so we can talk   
      radio.write( &got_time, sizeof(unsigned long) );              // Send the final one back.      
      radio.startListening();                                       // Now, resume listening so we catch the next packets.     
      Serial.print(F("Sent response "));
      Serial.println(got_time);  
   }
 }

/****************** Change Roles via Serial Commands ***************************/

  if ( Serial.available() )
  {
    char c = toupper(Serial.read());
    if ( c == 'T' && role == 0 ){      
      Serial.println(F("*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK"));
      role = 1;                  // Become the primary transmitter (ping out)
    
   }else
    if ( c == 'R' && role == 1 ){
      Serial.println(F("*** CHANGING TO RECEIVE ROLE -- PRESS 'T' TO SWITCH BACK"));      
       role = 0;                // Become the primary receiver (pong back)
       radio.startListening();
       
    }
  }

} // Loop
`

Output:
`
RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
printDetails
printDetails End
`

Alicetech, 2016-08-18:
I ended up fixing it by adding a function to RF24.cpp:

```
void printf_serial(const char* format, ...) 
{    
	char buffer[1000];

	va_list argptr;
	va_start(argptr, format);
	vsprintf_P(buffer, format, argptr);
	va_end(argptr);
	
	
	Serial.print(buffer);
}
```
And then adding this to RF24_config.h

```
#define printf_P printf_serial
```

I have no idea why nano wont printf_P to serial.... I don't even know if any Arduino can....

Avamander, 2016-08-19:
Why did you not add  "#include <printf.h>" ?

closed 2016-09-02 by TMRh20

patchido, 2016-09-16:
i did include printf.h wont work, and i dont really want to modify the rf24 library

LostStudent, 2018-04-17:
Im not getting any printDetails either on a replica uno, and cant seam to use it to send data, It works with the original maniacbug library though.  the above didnt fix the the issues im having. any tips?

RavikumarTulugu, 2021-06-29:
@Alicetech  thanks a ton, i ran into the same problem you did, i am using arduino pro mini , printDetails wont print any thing on to console, i used your code and it worked for me as well, i would request @TMRh20 to open the bug and fix it in the next official release.

2bndy5, 2021-06-29:
We're working on a better solution called RF24Log library.