nRF24/RF24 #279: Issues in Arduino pro mini

ru0, 2016-08-15:
Arduino pro mini connect the the pins of the NRF24L01(+)

compile Transfer.ino with Arduino  IDE 1.6.5.

PIN 	NRF24L01 	Arduino UNO 	
1 	GND 	       GND 	
2 	VCC 	               3.3V 	
3 	CE 	              digIO 7 	
4 	CSN 	              digIO 8 	
5 	SCK 	              digIO 13 	
6 	MOSI 	      digIO 11 	
7 	MISO 	      digIO 12 

the debug info like this,it seem not the right address,i try many times with different Lib,even i change the nRF24l01 modle,but it don't work.I don't know what the problem is.Can you help me?

STATUS		 = 0x7f RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0x7e7e7e7e7e 0x7e7e7e7e7e
RX_ADDR_P2-5	 = 0x38 0x38 0x38 0x3c
TX_ADDR		 = 0x7e7e7e7e7e
RX_PW_P0-6	 = 0x38 0x38 0x38 0x38 0x38 0x38
EN_AA		 = 0x3c
EN_RXADDR	 = 0x3c
RF_CH		 = 0x38
RF_SETUP	 = 0x38
CONFIG		 = 0x7e
DYNPD/FEATURE	 = 0x38 0x38
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_MIN

Avamander, 2016-08-15:
What exactly is the issue?

ru0, 2016-08-16:
@Avamander I bought the step-down IC and capacitors for more testing, i 'll tell you my trick.

ru0, 2016-08-19:
@Avamander Can u leave me your email address?i will send you my problem details.

Avamander, 2016-08-19:
It would be the best if you'd post your issue here, it might help someone
in the future.

ru0, 2016-08-20:
Connect the module to Arduino pro mini as bellow. 

[image]

[image]

I have already added capacitor of 10 uf between the pin 1 and pin 2 of the NRF2401+,and I download the RF24 library through the Library Manager of Aeduino IDE.I also download on github,but it’s the same.I have already tried the lib on https://github.com/maniacbug/RF24/.

[image]

This is the debug info.

STATUS		 = 0xff RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	 = 0x7f 0x7f 0x7f 0x7f
TX_ADDR		 = 0x7f7f7f7f7f
RX_PW_P0-6	 = 0x7f 0x7f 0x7f 0x7f 0x7f 0x7f
EN_AA		 = 0x7f
EN_RXADDR	 = 0x7f
RF_CH		 = 0xff
RF_SETUP	 = 0xff
CONFIG		 = 0xff
DYNPD/FEATURE	 = 0xff 0xff
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_MAX

 Initiating Extended Timeout Data Transfer Transfer complete at 10.00 KB/s 
 1000 of 1000 Packets Failed to Send
Data: .some strings...................

It can’t send data，the register address seem wrong.

The problem is similar to issues185,but They haven't been able to solve yet.
https://github.com/TMRh20/RF24/issues/185

Avamander, 2016-08-20:
Please install the pingpair sketch on two arduinos (don't modify anything else than what is told), does that work? Also, cycle the power of the arduinos after loading the sketch on them.

ru0, 2016-08-21:
@Avamander There is no necessary to do like you said,but i have already done that,Did you see that?
RX_ADDR_P0-1 = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5 = 0x7f 0x7f 0x7f 0x7f
TX_ADDR = 0x7f7f7f7f7f
The RX_ADDR_P0-1 egister address seem wrong.I just init the module
I tested the nRF24 module all i have on my pi(i have tow pi),It works fine.

Avamander, 2016-08-21:
That is the reason I told to use the default sketch, to see if your modifications are the issue.

ru0, 2016-08-22:
@Avamander no,i used the default sketch (Transfer.ino),i juest print the debug info.

Avamander, 2016-08-22:
As seen from one of your previous comment, you have changed definitely more
than just added debug print. Just take the default pingpair example, change
nothing unless told in the sketch.

ru0, 2016-08-22:
@Avamander ,that's just a example working on my pi.i mean that the debug info in the Arduino pro mini is not right,i really really do change nothing.

Avamander, 2016-08-22:
Okay, does the pingpair example work? You have to determine that the issue isn't the transfer example.

ru0, 2016-08-22:
@Avamander no,i tested the default  sketch  GettingStarted.ino and print debug info,is't the same
.
RF24/examples/GettingStarted
*** PRESS 'T' to begin transmitting to the other node
STATUS		 = 0x7f RX_DR=1 TX_DS=1 MAX_RT=1 RX_P_NO=7 TX_FULL=1
RX_ADDR_P0-1	 = 0xffffffffff 0xffffffffff
RX_ADDR_P2-5	 = 0xff 0xff 0xff 0xff
TX_ADDR		 = 0xff7fff7fff
RX_PW_P0-6	 = 0x7f 0x7f 0x7f 0x7f 0x7f 0x7f
EN_AA		 = 0x7f
EN_RXADDR	 = 0x7f
RF_CH		 = 0x7f
RF_SETUP	 = 0x7f
CONFIG		 = 0xff
DYNPD/FEATURE	 = 0xff 0xff
Data Rate	 = 1MBPS
Model		 = nRF24L01
CRC Length	 = 16 bits
PA Power	 = PA_MAX

ru0, 2016-08-22:
@Avamander i tested the minimum code ,it's the same.
----------------
#include <SPI.h>
#include "RF24.h"
#include <printf.h>

RF24 radio(7,8);

const uint64_t pipe = 0xE8E8F0F0E1LL;
 
void setup(void){
  Serial.begin(115200);
  printf_begin();
  
  radio.begin();
  radio.openWritingPipe(pipe);
  radio.printDetails();
}
 
void loop(void)
{
}

Avamander, 2016-08-22:
That was not what I meant for you to test. Do not change any pipes,
anything, unless the sketch says so.

ru0, 2016-08-22:
@Avamander The problem is not what i change but ... I don't know either.

kovax3, 2018-03-23:
tu a utiliser 5v votre rf24 ne fonction plus comme mois

closed 2019-05-10 by TMRh20