nRF24/RF24 #281: platformio exclusion of examples

ifilko, 2016-08-25:
/examples
/examples_linux
/pyRF24
/tests
should be in exclude section in library.json

zgoda, 2016-08-26:
Most libraries include examples in exported code. Moreover, this has any impact on package size only.

andreibosco, 2016-09-03:
@zgoda actually platformIO builds everything by default, including the examples, and right now `examples_linux` is causing build errors

jeanmatthieud, 2016-09-04:
Even without those folders, platformio still includes all /utility/* files, causing build errors

Avamander, 2016-09-04:
Pretty sure that was fixed in the last release?

andreibosco, 2016-09-04:
Tried yesterday and I'm still getting build errors on platformio.

jeanmatthieud, 2016-09-04:
Tried today with the last release (pio update)

TMRh20, 2016-09-05:
It seems these issues came up since https://github.com/TMRh20/RF24/pull/263  

Per https://github.com/TMRh20/RF24/pull/263/files it looks like the library.json should be reverted to:
```
-    "arch/ATTiny/*",
-    "arch/Due/*",
-    "arch/Teensy/*", 
```
instead of 
```
+    "utility/*",```

?

Avamander, 2016-09-05:
I'd go and ask for assistance from the platformio people, as it seems platformio has issues with compiling everything it gets it's hands onto.

zgoda, 2016-09-05:
CC @ivankravets @valeros

ivankravets, 2016-09-05:
PlatformIO 2.0 has a problem with Arduino libraries v1.0 (where `utility` folder is used). 

1. This issue is fixed in PlatformIO 3.0 (will be released this week)
2. Please switch to development version of PlatformIO. It should work http://docs.platformio.org/en/stable/installation.html#development-version

jeanmatthieud, 2016-09-05:
Thanks. I will wait next week

ivankravets, 2016-09-05:
@jeanmatthieud Please follow us on Twitter https://twitter.com/PlatformIO_Org and you will be notified about new releases.

closed 2016-11-10 by TMRh20