nRF24/RF24 #283: Question: How to improve delays of topology with a master node and multiple slaves?

juanmalunatic, 2016-08-31:
Greetings! I'm aware you guys are not running a support forum but I would really appreciate some quick pointers. 

I'm trying to implement a star topology (7 slaves, 1 master) with the nRF24L01+ and I've had moderate success so far: A master module polls each slave sequentially, sending 1 byte to the slave's pipe. The polled slave answers with an ACK packet that contains a 28 byte payload (read from a sensor). My goal is to achieve a stable 1.4 ms but my actual times oscillate between 1.3 and 1.6 ms (to be expected when using wireless I guess :) after setting RF24_2MBPS, RF24_PA_MAX and a 3 byte address.

If I could trim ~200us somewhere in the library (even if I lose a bit of reliability) I could "pad" every reading to achieve a stable framerate that suits my needs (with delays). Maybe my usage of the library is inefficient and there's a better way to solve this problem? Maybe I need to comment every single delay I find in the source files? (I haven't yet, tho :)

I'm using the Adafruit Feather M0 (SAMD21) as a platform and the attached code. Thanks A LOT for your hard work and advice.

Code:

```
// Includes omitted

/************* User config **********************/

// module 0 is the master node, anything else is a slave

byte module_id = 0;

/************************************************/

// module loop control
byte module_polling = 1; // starts with module1
byte total_slaves   = 7; // total slaves (shocking, I know)

RF24 radio(5,6); // CE,CSN pins on the RF24

// Byte used to poll by master
byte polling_byte = 1;

// Loop timing
unsigned long time_loop_start;

// One pipe address per slave

uint8_t addresses[][6] = {"1Addr", "2Addr", "3Addr", "4Addr", "5Addr", "6Addr", "7Addr", "8Addr"};
bool is_master;

void setup(void)
{
	delay(1000); // Delay the start just to see the first serial messages.

	// Set master flag for code readability
	if ( module_id == 0 )
	{
		is_master = true;
	}

	// Serial setup
	if( is_master )
	{
		Serial.begin(115200); // TO-DO: Well thought (minimum?) data rate
		
		if( debug_mode )
		{
			Serial.println(F("System startup."));
		};
	}

	// Start the radio
	bool radio_begin = radio.begin();
	
	if ( radio_begin && is_master )
	{
		Serial.println(F("RF24 began correctly."));
	};

	// Radio options must be activated after radio.begin
	radio.setDataRate( RF24_2MBPS  );     // Max speed
	radio.setPALevel( RF24_PA_MAX );      // Max power
	radio.setCRCLength( RF24_CRC_8 ) ;    // Min CRC
	radio.maskIRQ(1,1,1);                 // Turn off interrupts to pin
	radio.setAddressWidth(3); 			  // Shortest address
	radio.setChannel( 120 );              // Default is 76. Maybe helps with BT interference.
	
	radio.setRetries(1, 0); // Minimum combination that works with my Hw.

	// Enable ACK payloads
	radio.enableAckPayload();          // Enable ACK payloads in pipes 0 and 1.
	radio.enableDynamicPayloads();     // Enable ACK payloads in all pipes (both lines are required!)
	
	delay(1000);

	// Initial pipe set-up

	if ( is_master )
	{
		radio.openReadingPipe(1, addresses[7]);  // Master's address is [7] on the list
		radio.openWritingPipe( addresses[0] );   // Master begins polling 1Addr
	} 
	else 
	{
		// Always read on pipe 1
		// Example: 1Addr (addresses[0]) is used for broadcasting to module 1
		radio.openReadingPipe(1, addresses[module_id - 1] );

		//Slaves always write to the master's address
		radio.openWritingPipe( addresses[7] );
	}
	
	radio.startListening(); // Necessary for all.

	if( is_master )
	{
		radio.stopListening();      // Master stops listening (listening is the default slave behaviour).
	}

	// IMU setup just for slaves
	if( !is_master )
	{
		// Check if the device was initialized properly.
		if( !bno.begin() )                       
		{
			// sensor stuff
		}
		else
		{
			// moar sensor stuff
		}
	}

}

void loop(void)
{

	// Times the master loop.
	if( is_master )
	{
		time_loop_start = micros();
	}
	
	/////////////////////////////////////////////

	if (is_master) {
		
		/****************************************/
		//         Master module's loop
		/*****************************************/

		// Placeholder to receive the payload.
		// Length here must always be max_index+1 (27+1)!
		byte ack_payload[28];  
		  
		//Module iteration
		// 200 us
		radio.openWritingPipe( addresses[module_polling - 1] ); // Switch target
		
		// Diagnose and timeout
		unsigned long polling_begins = micros();
		bool timeout  = false;
		bool write_ok = false;
		
		// Writes the polling byte to module N

		// write:     910 on fail, 610 on succ [us]
		write_ok = radio.write(&polling_byte, 1);
		
		// If sent a tx successfully a slave must respond.
		if ( write_ok )
		{
			//while( (! radio.available()) && (! timeout ) ) for reception timeouts
			while( radio.available() )
			{
				// Read the payload
				// 200 us
				radio.read( ack_payload, 28 );   // if not receiving an array &dereference the var
				
				// Output the payload to the serial port
				Serial.write(ack_payload, 28); // Write the received payload as bytes. Neat!
				Serial.println(""); // Provisional line break to differentiate
				
			}
			
		} //end if radio.write 
		else
		{
			// Error writing to the destination
			// Retries are disabled and the frame rate depends on the network speed).
			Serial.print(module_polling, DEC);
			Serial.print(" is kill.");
			Serial.println("");
		}

		// Iterate to next module
		if ( module_polling < total_slaves )
		{
			module_polling = module_polling + 1;
		}
		else
		{
			module_polling = 1;
		}

		Serial.print(".TL: ");
		Serial.print( micros() - time_loop_start );
		Serial.println(F(""));
		
	} // end if master module
	else
	{

		/****************************************/
		// Slave module's loop
		/*****************************************/

		byte pipeNo, gotByte;
		byte frame_imu[28]; // same as above, declaration contains the length (28 bytes)
			
		while ( radio.available(&pipeNo) ) //pipeNo stores the pipe used in the tx
		{  
			// Dummy package arrives
			// First ACK after power on is sent w/o payload
			radio.read( &gotByte , 1 );

			// Dummy mode: BNO055 not connected 
			// Read an actual frame from the sensor
			createFrame( module_id, frame_imu );	

			// Finally, piggyback the data onto the ACK
			// Notice how this setups the NEXT transmission only (not an issue).
			radio.writeAckPayload(pipeNo, frame_imu, 28 );

		};

	};

};
```

TMRh20, 2016-09-02:
Ack payloads are largely handled internally, so the simplest way to minimize response times and ensure more consistency is by using interrupts. 

See http://tmrh20.github.io/RF24/pingpair_irq_simple_8ino-example.html
and http://tmrh20.github.io/RF24/pingpair_irq_8ino-example.html

Notes:
1. The slave module can also pre-load an ack payload, so the first response is not blank.
2. This code ``` Serial.print(".TL: ");  Serial.print( micros() - time_loop_start ); ``` will include the time to Serial.print ".TL: " characters. Typically, the timing should be calculated prior to printing anything if you want accuracy.
ex: ``` time_loop_start = micros() - time_loop_start;    Serial.print(".TL: ");    Serial.print(time_loop_start ); ```
```

MisterDudeManGuy, 2016-09-30:
From a system design standpoint, I'd never want to design such that I'm are so very dependent upon the delivery time of messages from remote nodes being the same.  That is a recipe for frustration because it is not possible in the general case since you cannot control the quality of the airwaves and thus the likelihood of retries (IE more time for some packets).  Varying device state at all of the nodes also assures some variance in timing.  Wireless systems should be loosely coupled so that normal communications delays and variances are handled transparently by the design.

If 300us of timing skew in responses is a problem for your system, then I'd tend to revisit base assumptions in system design before doing anything else.  I would not, for example, assume this kind of stringent timing consistency on a even a hard-wired network - nor would I revise underlying libraries like my TCP stack if my wired network did not deliver the timing my design required.  That is because network collisions, device state, and 'other stuff we can't control' all conspire to throw in some timing skew.

Just my two cents worth. Have fun!

ma4185, 2016-10-13:
Hello, which includes did you use? Because when I try to compile the code, I get the error message that debug_mode, bno and createFrame were not declared in this scope.

Avamander, 2016-10-13:
@ma4185 what platform are you compiling for? By default it should require
no defines other than what are there.

juanmalunatic, 2016-10-13:
First of all, a big "thank you" to everyone that has answered so far.

@ma4185 

CreateFrame is an external cpp file, it basically startis an i2c transfer and fetches data from a sensor. If you're adapting this for your own use, replace that line for whatever action you want your slave to perform. Bno is another sensor-specific library and it's used to initialize the sensor. debug_mode is a boolean that's not included, but also not critical. Short answer: these parts don't have anything to do with the RF24 library :)

@TMRh20 

Thanks for the advice, I haven't actually rewritten the whole thing using interrupts because removing the unnecessary Serial.prints solved all my timing requirements (I know, shame on me for not testing this previously). I also reduced my dropped packages significantly by switching to RF24_CRC_16, as you suggested in some other answer.

@MisterDudeManGuy 

Thanks for the insight, your design recommendations are very sensible. This design assumes the best case scenario (as I have to meet some latency requirements) but you're absolutely right in all your observations. In the current version the master just reports a "blank" packet if timing requirements are not met or if any kind of error occurs (not really robust). I'm not really sure of how to overcome this with a Wireless system without sacrificing the "real-time" aspect of the data. 

I've changed some things around since the time this was posted, including a basic way to gather some stats (although I suspect my error detection logic is iffy). If you guys think it might be useful for someone, I'll upload it.

ma4185, 2016-10-26:
Hello, i took your code and change it a little bit for my project.
There is a Raspberry, that gets the master_role or role_ping_out and at the moment three Arduinos for slave or role_pong_back. The Raspberry should write one Arduino after the other.
Now I have the problem, that the programms start, but they don't find each other. Do I have to change the module_polling for each Arduino?

Arduino code:

#include <SPI.h>
#include "nRF24L01.h"
#include "RF24.h"
#include "printf.h"

RF24 radio(7, 8); // CE,CSN pins on the RF24

byte addresses[][6] = {"1Addr", "2Addr", "3Addr", "4Addr"};

typedef enum { role_ping_out = 1, role_pong_back } role_e;                 // The various roles supported by this sketch
const char* role_friendly_name[] = { "invalid", "Ping out", "Pong back"};  // The debug-friendly names of those roles
role_e role = role_pong_back;                                              // The role of the current running sketch

byte module_polling = 1; // starts with module1
byte total_slaves   = 3; // total slaves (shocking, I know)
byte counter = 1;
byte module_id=0;

void setup() {

  delay(1000);
  Serial.begin(115200);

  printf_begin();
  printf("\n\rRF24/examples/MyStern3/\n\r");
  printf("ROLE: %s\n\r", role_friendly_name[role]);
  printf("*** PRESS 'T' to begin transmitting to the other node\n\r");

  radio.begin();
  radio.enableAckPayload();
  radio.enableDynamicPayloads();
  radio.setRetries(1, 15);

  radio.setDataRate( RF24_2MBPS  );     // Max speed
  radio.setPALevel( RF24_PA_MAX );      // Max power
  radio.setCRCLength( RF24_CRC_8 ) ;    // Min CRC
  radio.maskIRQ(1, 1, 1);               // Turn off interrupts to pin
  radio.setAddressWidth(3);             // Shortest address
  radio.setChannel( 120 );              // Default is 76. Maybe helps with BT interference.

  radio.openWritingPipe(addresses[3]);      // Since only two radios involved, both listen on the same addresses and pipe numbers in RX mode
  radio.openReadingPipe(1, addresses[module_id-1]);   // then switch pipes & addresses to transmit.
  radio.startListening();

  radio.powerUp();
  radio.printDetails();

}

void loop (void) {

  if (role == role_ping_out) {

    delay(900);
    radio.stopListening();
    printf("Bitte anderes Modul waehlen.\n\r");
  }

  if (role == role_pong_back) {

    byte pipeNo, polling_byte;

    while (radio.available(&pipeNo)) {

[... 140 lines cut ...]
		if ( module_polling < total_slaves )
        {
            module_polling = module_polling + 1;
            delay(900);
            printf("Modul %d wird angesprochen. \n\r", module_polling);
            
        }
        else
        {
            module_polling = 1;
            delay(900);
            printf("Kreislauf beginnt von Neuem. \n\r");
           }
			
	}
			
	if (role == role_pong_back) {
		
		delay(900);
		printf("Bitte den anderen Modus nutzen. \n\r");
				
	}
	
}	
}

closed 2016-11-10 by TMRh20