nRF24/RF24 #285: pingpair_ack example does not work at 250kbps

SergioUno, 2016-09-15:
Hi, 
I think this is a bug in a library.
I am usint Arduino IDE1.6.11. and the latest version of the library (1.1.7)

I added in the Setup section this line
  radio.setDataRate(RF24_250KBPS); 
and it totally block the ping. 

However,   
radio.setDataRate(RF24_1MBPS); 
OR
radio.setDataRate(RF24_2MBPS); 
both works fine.

See details below

```
RF24/examples/pingpair_ack/

ROLE: Pong back
*** PRESS 'T' to begin transmitting to the other node
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0x544d52687c 0xabcdabcd71
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0x544d52687c
RX_PW_P0-6	 = 0x01 0x01 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x64
RF_SETUP	 = 0x25
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x03 0x06
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_HIGH
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending 1 as payload. failed.
Now sending 1 as payload. failed.
Now sending 1 as payload. failed.
Now sending 1 as payload. failed.
Now sending 1 as payload. failed.
```

SergioUno, 2016-09-16:
I checked that GettingStarted example works with the same modules at 250 kbps, so the hardware is plus version and it is ok.

Maybe the problem in this function which is sensitive to 250 kbps?

```
uint8_t RF24::flush_tx	(	void 		)	
Empty the transmit buffer. This is generally not required in standard operation. May be required in specific cases after stopListening() , if operating at 250KBPS data rate.
```

TMRh20, 2016-09-16:
Check the setRetries config vs min values at 250K

SergioUno, 2016-09-16:
It is default:
`  radio.setRetries(0,15);                 // Smallest time between retries, max no. of retries`

closed 2016-09-19 by TMRh20

Syrex-o, 2021-07-02:
Just an additional informer for other people with the same problems:
My problem was not fixed by changing the retires to: `radio.setRetries(5,15);`

In order to get `RF24_250KBPS` working, in frequent test conditions, I had to put an additional standby period on the receiving Arduino. This is the only way for the Raspberry to receive the manual Ack (at least for me).

`radio.stopListening(); `
`radio.txStandBy(100); `
`radio.writeFast(&payload, bytes);`
` radio.txStandBy(50);`
` radio.startListening();`

2bndy5, 2021-07-02:
The _pingpair_ack_ example has been replaced with the new _AcknowledgementPayloads_ example as of v1.3.11 of the RF24 lib. BTW, the _ManualAcknowledgments_ example is fundamentally different from _AcknowledgementPayloads_ example (which uses the radio's auto-ack feature).

Syrex-o, 2021-07-03:
@2bndy5 quite fast answer :)

But its more about the manual ack for `RF24_250KBPS `.
The example is not important :)

2bndy5, 2021-07-03:
I just wanted to add clarity for passersby. I appreciate you posting your solution (even though the original issue was about the auto-ack example).