nRF24/RF24 #287: RF24 and Attiny84

DoctorZIP, 2016-09-24:
Hi,
I trying to get worked RF24 library on attiny84 over the month but vain. Example from this source code doesn't works for me. Attiny84 connected to nrf24l01+ as described in in-comment documentation. Also SPI sequenses were not detected using logic analyser. Plese help me to found right way to solve my problem.

SashkaCosmonaut, 2016-09-24:
I use this library with ATtiny85 and Arduino IDE. But I don't know is it much different from your version of ATtiny. If it will help, I can give you my code as an example.

DoctorZIP, 2016-09-24:
@SashkaCosmonaut It would be great.

SashkaCosmonaut, 2016-09-25:
http://arduino.stackexchange.com/questions/18163/using-attiny85-nrf24l01-dht2-cant-get-data-from-dht22/ There is my question, where I described a problem of my project (that is already solved) and there are all my code examples. But your problem my be caused not only by code and the circuit must be correct and the programming process too.

DoctorZIP, 2016-09-25:
@SashkaCosmonaut Thank you for device example it will be very useful for me in the future, but not now.

Here is my sketch code:
`#define __AVR_ATtiny84__

#include <RF24.h>
#include <nRF24l01.h>

#define CE_PIN 7
#define CSN_PIN 8

byte addresses[][6] = {
  "1Node","2Node"};

RF24 radio(CE_PIN, CSN_PIN);
int data = 0;
void setup(void)
{
  delay(500);
  radio.begin();
  delay(2);
  radio.setChannel(9);
  radio.setAutoAck(1); // Ensure autoACK is enabled
  radio.setRetries(15,15); // Max delay between retries & number of retries
  radio.openWritingPipe(addresses[1]); // Write to device address '2Node'
}

void loop(void) {
  data += 1;
  radio.write(&data, sizeof(data));
  delay(100);
}`

SashkaCosmonaut, 2016-09-25:
Can you tell me please some more: 

- What device and code do you use to receive messages from your ATtiny?
- How often do you check data on your receiver? Somehow I faced a problem when I sent data very often and a receiver wasn't able to read data.
- How do you program your ATtiny84? Maybe it is a silly question but are you sure that this code really works (for example, did you connect a led after every data sending)?
- How do you connect nrf24l01+ to your ATtiny? Do you connect pins directly or use some circuit?
- Why do you use setChannel function? Is it required? I am asking because I never used it.

DoctorZIP, 2016-09-25:
@SashkaCosmonaut 
The code listed above is start point of my development. In completely done project data will be transferred every 8 seconds. I using "China" USBasp programmer that works well and blink sketches also works fine. SPI was tested by connecting to Logic Analyzer like [this](http://we.easyelectronics.ru/Pika4u56/saleae-logic-ili-poddelka-dyadyushki-lyao.html) Nrf24l01+ data pins connected directly to MCU and power supply pins connected through AMS1117-3.3 voltage regulator with additional capacitors. First connection scheme was used from [this](https://github.com/TMRh20/RF24/blob/master/examples/rf24_ATTiny/rf24ping85/rf24ping85.ino) official examples:
```
ATtiny24/44/84 Pin map with CE_PIN 8 and CSN_PIN 7
	Schematic provided and successfully tested by Carmine Pastore (https://github.com/Carminepz)
                                  +-\/-+
    nRF24L01  VCC, pin2 --- VCC  1|o   |14 GND --- nRF24L01  GND, pin1
                            PB0  2|    |13 AREF
                            PB1  3|    |12 PA1
                            PB3  4|    |11 PA2 --- nRF24L01   CE, pin3
                            PB2  5|    |10 PA3 --- nRF24L01  CSN, pin4
                            PA7  6|    |9  PA4 --- nRF24L01  SCK, pin5
    nRF24L01 MOSI, pin7 --- PA6  7|    |8  PA5 --- nRF24L01 MISO, pin6
                                  +----+ 
```
also I tested CE and CSN on 7 and 8 pins - no results.

setChannel was used beacuse it mentioned in documentation, it's a good idea try to remove it.
**But**
I tryied to use Attiny85 and you code and I see SPI seequences on logic analyzer. So attiny84 SPI not working with this.

SashkaCosmonaut, 2016-09-25:
Ok. Excuse me, but I have another silly question: there already is a couple of issues about ATtiny84 (https://github.com/TMRh20/RF24/issues/236, https://github.com/TMRh20/RF24/issues/93). They didn't helped you, aren't they?

DoctorZIP, 2016-09-25:
Thanks for hints. I'll check it for possible solutions.

closed 2016-11-10 by TMRh20