nRF24/RF24 #288: Network Default Address (clarification please)

luudee, 2016-09-25:
Hi !

First of all I would like to say you are doing an absolutely excellent job developing this library !
I am working on a large project that will deploy thousands of nodes over a wide area !

I understand the network topology, and you state in the online description that 05555 would
be the highest address achievable, supporting 3,125  nodes. However, it appears you are
using 04444 as the network "default address".  I am not sure I would really use the 3,125
nodes, but I can see how in my case 04444 could belong to a valid node.

Is it ok to change this value to 07777 as there would be no real node with that address ?
I am a bit confused what the purpose if this address is anyway. Could you be so kind and
comment on this ?

Also, may I humbly suggest to not hard code values like 04444 but rather use defines for
this ? It would make future changes and upgrades so much easier. :)

Many Thanks,
rudi

TMRh20, 2016-09-27:
> Is it ok to change this value to 07777 as there would be no real node with that address ?

Good question.

The 04444 address is from RF24Mesh development. New nodes use it while requesting addresses. It should be changed to an unused address like 07777, but the details need to be worked out & tested.

> Also, may I humbly suggest to not hard code values like 04444 but rather use defines for
this ?

Yeah, it is defined in RF24Mesh, but maybe should be defined in network layer... the whole thing needs a review.

closed 2019-05-10 by TMRh20

2bndy5, 2021-07-02:
> Is it ok to change this value to 07777 as there would be no real node with that address ?

I was going to ask the same thing, but we need to use an address that would pass `is_valid_addr()` check (0777 will not). So, the default address needs to only have digits 1-5 (0 is reserved for master & 0100 is reserved for multicasting). I like the idea of using 07777, but it would require adding a conditional statement to `is_valid_addr()`.