nRF24/RF24 #289: Enhancement Request: Watch Dog Timer support (indirect)

luudee, 2016-09-25:
Hi !

would it be possible to have a way to add a call back function in the code that
takes a while to complete ? In several places, you are waiting for the radio or
other events in hard loops. Any implementation that uses watch dog times, will
need to make the timeout unreasonably high ...

If users had an option to add a call back routine, that would reset the watchdog
timer, that would be really helpful !!!

Many Thanks,
rudi

TMRh20, 2016-09-27:
> would it be possible to have a way to add a call back function in the code that
takes a while to complete ?

Like using interrupts? http://tmrh20.github.io/RF24/pingpair_irq_simple_8ino-example.html

luudee, 2016-09-27:
No, this would just give me an interrupt when the radio has something.

Resetting (feeding) the watchdog needs to be done within a certain interval,
or the entire system will be reset. So a real overload for all the delay calls
in the code would be really helpful. It would also help if one would want to
use the Mesh within an RTOS, as t hey require a special "delay" function to be
called.

Many Thanks,
rudi

TMRh20, 2020-08-16:
Not gone, not forgotten, apologies for almost taking 4 years, but feature added to RF24Mesh. Thanks for the suggestion.

The planned documentation is as follows:
  
  * RF24Mesh ID and Address lookups as well as address renewal can take some time.
  * Set a callback function to enable additional processing while the mesh is working
   ```
   void myCallbackFunction(){
     someValue = someOtherValue
   }
   mesh.setCallback(myCallbackFunction);
   ```
  ``` @param meshCallback The name of a function to call ```
<br>
The changes are committed to master, and I will be doing a release, probably for all the RF24 libs, pretty soon due to many bug fixes and changes.

Closing, please advise if you are still around and there are any problems.

closed 2020-08-16 by TMRh20