nRF24/RF24 #292: What's different between a object pointer and a object?

lukenothing, 2016-10-01:
Hello, I found a strange thing when I used the RF24 class.

This is my test code as follow.  I try to use a RF24 object pointer instead of a RF24 object to run, but it will restart the arduino uno again, again and again. After it calls any RF24 functions except 'available()', it will print endless lines of 'Receiver Starting...' on the serial monitor window, which means that arduino uno reenter the setup repeatedly.

Perhaps it is not a BUG, and that is just the wrong way I use RF24 ?

`#include <RF24_config.h>
#include <RF24.h>
#include <nRF24L01.h>

RF24 *radio;

byte addresses[][6] = { "1Node","2Node" };
//const uint64_t pipe = 0x12345678ABLL;

void setup() {

	radio = new RF24(9, 10);

	Serial.begin(115200);
	Serial.println("Receiver Starting...");

	radio->begin();

	radio->setPALevel(RF24_PA_LOW);

	radio->setRetries(15, 15);

	radio->setChannel(25);

	radio->setPayloadSize(sizeof(unsigned long));

	radio->openWritingPipe(addresses[1]);
	radio->openReadingPipe(1, addresses[0]);

	radio->startListening();

	radio->printDetails();
}

// the loop function runs over and over again until power down or reset
void loop() {
	if (radio->available())
	{
		unsigned long got_time = 0;
		while (radio->available())
		{
			radio->read(&got_time, sizeof(unsigned long));

			// Spew it
			//Serial.print("Got payload ...");
			//Serial.print(got_time, DEC);

			// Delay just a little bit to let the other unit
			// make the transition to receiver
			//delay(20);
		}

		if (got_time != 0) {
			Serial.print("Got payload ...");
			Serial.print(got_time, DEC);
		}
		else {
			Serial.println("got_time = 0");
		}

		// First, stop listening so we can talk
		radio->stopListening();

		// Send the final one back.
		if (radio->write(&got_time, sizeof(unsigned long)))
			Serial.println(F(". Sent response OK."));
		else
			Serial.println(F(". Sent response failed."));

		// Now, resume listening so we catch the next packets.
		radio->startListening();
	}
}`

lukenothing, 2016-10-01:
Oh, I guest it's a language problem ? I use 'Arduino IDE for Microsoft Visual Studio' in vs2015, the processing syntax does not support '->' ? I'm not sure that, because vs can successfully compile and link.

lukenothing, 2016-10-02:
Here is the code which could be running sucessfully.

	#include <Arduino.h>
	#include <RF24_config.h>
	#include <RF24.h>
	#include <nRF24L01.h>

	RF24 radio(9, 10);

	byte addresses[][6] = { "1Node","2Node" };

	void setup() {

		//radio = new RF24(9, 10);

		Serial.begin(115200);
		Serial.println("Receiver Starting...");

		radio.begin();

		radio.setPALevel(RF24_PA_LOW);

		radio.setRetries(15, 15);

		radio.setChannel(25);

		radio.setPayloadSize(sizeof(unsigned long));

		radio.openWritingPipe(addresses[1]);
		radio.openReadingPipe(1, addresses[0]);

		radio.startListening();

		radio.printDetails();
	}

	// the loop function runs over and over again until power down or reset
	void loop() {
		if (radio.available())
		{
			unsigned long got_time = 0;
			while (radio.available())
			{
				radio.read(&got_time, sizeof(unsigned long));

				// Spew it
				//Serial.print("Got payload ...");
				//Serial.print(got_time, DEC);

				// Delay just a little bit to let the other unit
				// make the transition to receiver
				//delay(20);
			}

			if (got_time != 0) {
				Serial.print("Got payload ...");
				Serial.print(got_time, DEC);
			}
			else {
				Serial.println("got_time = 0");
			}

			// First, stop listening so we can talk
			radio.stopListening();

			// Send the final one back.
			if (radio.write(&got_time, sizeof(unsigned long)))
				Serial.println(F(". Sent response OK."));
			else
				Serial.println(F(". Sent response failed."));

			// Now, resume listening so we catch the next packets.
			radio.startListening();
		}
	}

lukenothing, 2016-10-02:
Haha, perhaps I have already addressed the problem, which is function of 'printDetails()'.

When I comment this call, both of codes run well.

lukenothing, 2016-10-02:
It seems that printf() has some problem? #189

campenr, 2016-11-08:
Does it work if you add the include for ``print.f`` and add the line ``printf_begin();`` somewhere during your setup?

closed 2019-05-10 by TMRh20