nRF24/RF24 #293: error : bad value (cortex-a7) for -mtune switch

PaoloElMarinelo, 2016-10-01:
Hi !
I'm newbie on Linux and I'm discovering Python,
and I love it !!

Can I have your help please,
Trying to install RF24 lib on my Rpi 2.
After cloning RF24, I have this error, what could I do ?

cc1plus: error: bad value (cortex-a7) for -mtune switch
Makefile:40: recipe for target 'RF24.o' failed

`pi@raspberrypi ~/RF24 $ make
Makefile:17: Makefile.inc: No such file or directory
[Running configure]
[SECTION] Detecting arm compilation environment.
  [OK] arm-linux-gnueabihf-gcc detected.
  [OK] arm-linux-gnueabihf-g++ detected.
[SECTION] Detecting target machine.
[OK] machine detected: SoC=BCM2836, Type=RPi2, CPU=armv7l.
[SECTION] Detecting DRIVER
  [OK] DRIVER detected:RPi.
[SECTION] Detecting OS.
  [INFO] OS detected:LINUX.
[SECTION] Preparing configuration.
[SECTION] Saving configuration.
[OK] Finished.
arm-linux-gnueabihf-g++ -fPIC -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
cc1plus: error: bad value (cortex-a7) for -mtune switch
Makefile:40: recipe for target 'RF24.o' failed
make: *** [RF24.o] Error 1
`
Thank you very much for your help.
Paolo

PaoloElMarinelo, 2016-10-04:
Hi !
It is a problem of GCC ?
it's my Gcc  didn't understand this compiler flag : -mcpu=cortex-a7 ?
Thanks for little help.

Paolo

TMRh20, 2016-10-08:
Is your RPi all up-to-date?

``` sudo apt-get update; sudo apt-get dist-upgrade; sudo rpi-update ```

PaoloElMarinelo, 2016-10-13:
Yes, he is up-to-date and it makes no difference...
So I reinstalled Raspbian and now it works. :-)

closed 2016-10-13 by Avamander

lexotero, 2016-10-14:
I am having the same problem. I will try to run the command above and "make install" again. I will let you know if this solves the problem.

lexotero, 2016-10-14:
After running the command, the problem persists. I don't want to reinstall Raspbian. Any idea?

Many thanks and congrats for your job!!

PaoloElMarinelo, 2016-10-15:
I don't know if it's a good idea but you can try remove the option `-mtune=...` between the lines 151 to 169 in the "configure" file.
https://github.com/TMRh20/RF24/blob/master/configure

lexotero, 2016-10-15:
Yes, I was thinking about that but for sure this will cause another error xD

I'll try, many thanks!

Justin248, 2016-10-16:
Hi Lexotero,
i am having the same issue like you but still struggling. Could you fix it

lexotero, 2016-10-16:
No, but really I don't what to do, It is a failure trying to compile the library, then I prefer not to modify anything, if the solution makes the library work slower or no appropriately, I prefer to use the first computer principle "shutdown and switch on again". I'll reinstall the SO on my raspberry to check if maybe the problem was something with another modification I did previously, and if it is still not working, I will try to implement the library my self.

lexotero, 2016-10-17:
I have reinstalled the Raspberry and now the compilation passes. I don't know what was happening but now is working.

myTonino, 2016-10-17:
You are not alone. I just went though a similar thing. Compilation revealed an "internal compiler error".  Uninstalling/Reinstalling the compiler tools did not fix that. I investigated a full weekend and ended up reinstalling the Raspberry from scratch. Now everything works again as expected. Not sure what brought it to that broken state. Everything else worked fine. Hm.

lexotero, 2016-10-17:
Now I can't make the Mesh work, but that is a different problem! I try to run the mesh master example, it prints the configuration but nothing happens and the arduinos are trying to renew the address.
Well, many thanks for the help, even if was just for the advice of reinstalling the OS.

lexotero, 2016-10-17:
Working! Wire connection problem

Justin248, 2016-10-22:
Thanks all for the advice