nRF24/RF24 #297: Does radio sensibility of nRF24L01+ differ from RaspberryPi and ESP8266-12 ?

Slartitorto, 2016-10-14:
I built a system with about ten devices with Attiny84 and nRF24L01+ (simple version) that send temperature every 5 minutes from different places in my house; they send data to a receiver with a RaspberryPi and a nRF24L01+ with external antenna (24L01+PA+LNA SMA module).
The receiver is connected to the internet via ethernet cable to my internet router and stores data in a remote database. This works fine since 2 years without any problem and no datagram is lost. Some of the devices are in rooms far 15-20 meters from the receiver, often 3-4 wall in the meddle. Some devices are in my freezer, so i can monitor the temperature inside.
One of these devices is in a steel box: initially the receiver was receiving data from this one only once every 2-3 times, so I applied the antenna modification as describe here (http://www.instructables.com/id/Enhanced-NRF24L01/?ALLSTEPS). Now, also data from this device are received without any packet lost.

I’m trying to remake the receiver in order to obtain router with a ESP8266-12 connected via wifi and cheaper, for some of my friends. I made the prototype with the same nRF24l01 module with external antenna used in the Raspberry receiver (24L01+PA+LNA SMA module) and it works, but some of the remote sensors are not received every times they sent data: i.e. the freezer devices are received 7-8 times every 10, and the same the one in the steel box; also one sensor outside my house, but not the others that are received perfectly with no packet lost.

The software is substantially the same, the nRF24L01+ module is the same, the position is the same; I don’t think there are problems with wifi radio frequency, because the internet router with integrate AP is close to the receiver in both the cases, either the RaspberryPi receiver and the ESP8266 receiver.
The power supply is 3.3V supplied from the Raspberry in that case, supplied from a LE33 in the ESP receiver, both well filtered with good capacitor.

I don't understand why the ESP receiver loses some datagrams only from the sensor more far or more shielded.
I would not turn on the autoacknoledge flag and retries on the sensors in order to save battery (I suppose if the senor sends data more times, it consumes more energy)

In both cases (Raspberry and ESP receiver) I use:

  radio.begin();
  radio.enableDynamicPayloads();
  radio.setAutoAck(1);
  radio.setRetries(15,15);
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_MAX);
  radio.setChannel(76);
  radio.setCRCLength(RF24_CRC_16);
  radio.openReadingPipe(1,0xF0F0F0F0E1LL);
  radio.stopListening();
  radio.startListening();

This is the printDetails() output:

================ SPI Configuration ================
CSN Pin  	 = CE0 (PI Hardware Driven)
CE Pin  	 = Custom GPIO25
Clock Speed	 = 8 Mhz
================ NRF Configuration ================
STATUS		 = 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	 = 0xe7e7e7e7e7 0xf0f0f0f0e1
RX_ADDR_P2-5	 = 0xc3 0xc4 0xc5 0xc6
TX_ADDR		 = 0xe7e7e7e7e7
RX_PW_P0-6	 = 0x00 0x20 0x00 0x00 0x00 0x00
EN_AA		 = 0x3f
EN_RXADDR	 = 0x02
RF_CH		 = 0x4c
RF_SETUP	 = 0x27
CONFIG		 = 0x0f
DYNPD/FEATURE	 = 0x3f 0x04
Data Rate	 = 250KBPS
Model		 = nRF24L01+
CRC Length	 = 16 bits
PA Power	 = PA_MAX

In the sensor with attiny84 I use:

 radio.begin();
 radio.setPALevel(RF24_PA_MAX);
 radio.setChannel(0x4c);
 radio.setDataRate(RF24_250KBPS);
 radio.openWritingPipe(0xF0F0F0F0E1LL);
 radio.enableDynamicPayloads();

Any idea ?

Avamander, 2016-10-15:
You can try stabilizing the power supply by adding electrolytic capacitor(s) to the 3.3V rail. Usually 1000uF soldered to the GND and 3.3V pins of the radio. To filter high-frequency noise try adding 1uF ceramic capacitor to the 3.3V bus. Another thing you can do is disable the ESP8266's WiFi radio or something, as both work on the same frequency it might also add interference.

Slartitorto, 2016-10-18:
Thanks a lot,
I've found the cause of the problem: some datagram were received, but I didn't see on the receiver app. This because the same code works differently between Raspbian (c++) and ESP8266: the declaration:
`char receivePayload[32]="";
`
inside the loop cicle, but outside and above the
`while (radio.available())
`
This works fine on Raspberry, but not on ESP8266. For some reason the same code on ESP8266 leaves some of the last bytes of the variable receivePayload "dirty" and the software doesn't show certain datagrams.
I've moved the statement:
`char receivePayload[32]="";
`
just below:
`while (radio.available()) 
`
Now all is working fine.

I'm not a good software writer, so its easy to make some mistake.
Thanks again.

This is the simplified code on Raspberry:

```
// nRF24L01+ on RaspberryPi
// code for data RX
// connect:
// nrf24L01:     1   2   3   4   5   6   7
// RaspberryPi:  6   1   22  24  23  19  21

#include <cstdlib>
#include <iostream>
#include <RF24/RF24.h>

RF24 radio(RPI_V2_GPIO_P1_22, RPI_V2_GPIO_P1_24, BCM2835_SPI_SPEED_8MHZ);

void setup(void)
{
  // init radio for reading
  radio.begin();
  radio.enableDynamicPayloads();
  radio.setAutoAck(1);
  radio.setRetries(15,15);
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_MAX);
  radio.setChannel(76);
  radio.setCRCLength(RF24_CRC_16);
  radio.openReadingPipe(1,0xF0F0F0F0E1LL);
  radio.startListening();
}

void loop(void)
{
  // *** 32 byte character array is max payload
  char receivePayload[32]="";

  // *** Sleep 20 ms
  usleep(20000);

  while (radio.available())
  {
    // *** Read from radio until payload size is reached
    uint8_t len = radio.getDynamicPayloadSize();
    radio.read(receivePayload, len);
      
      char wget_command[256];
      sprintf(wget_command,"wget -q -b -O /dev/null http://www.myserver.eu/datasink.php?data=%s",receivePayload);
      system(wget_command);
  }
}

int main(int argc, char** argv)
{
  setup();
  while(1)
  loop();
  return 0;
}

```

And this is the one for ESP8266

 ```
// * nRF24L01+  ESP8266
 // * VCC        VCC
 // * CE         GPIO4
 // * CSN/CS     GPIO15
 // * SCK        GPIO14
 // * MISO       GPIO12
 // * MOSI       GPIO13
 // * GND        GND

 // use NODE MCU (ESP-12E) as Arduino type

#include <ESP8266WiFi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <WiFiClient.h>
#include <RF24.h>

#define LED 5
#define BUTTON 16

// Remote site information
const char http_site[] = "www.myserver.eu";
const int http_port = 80;

// Initialize RF24 library
RF24 radio(4,15);

void setup() {
  Serial.begin(115200);
  pinMode(LED, OUTPUT);
  pinMode(BUTTON, INPUT_PULLUP);

  WiFiManager wifiManager;
  wifiManager.setAPCallback(configModeCallback);
  wifiManager.autoConnect("BARS_AccessPoint");
  // put RED_LED off after configuration
  digitalWrite(LED, LOW);

  yield();
  restartRadio(); 
}

void loop() {

  delay(20);
  if (radio.available()) {
    while (radio.available()) {
      yield();
      uint8_t len = radio.getDynamicPayloadSize();
      char receivePayload[32]="";
      radio.read(receivePayload, len);

        digitalWrite(LED, HIGH);

        // Send data to sink

        WiFiClient client;
        if (!client.connect(http_site, http_port) ) {
          blink_error();
        } else {
          client.print("GET /datasink.php?data=");
          client.print(receivePayload);

          client.println(" HTTP/1.1");
          client.print("Host: ");
          client.println(http_site);
          client.println("Connection: close");
          client.println();
        }
        digitalWrite(LED, LOW);     
    }
  }
}

void restartRadio(){
  yield();
  radio.begin();
  radio.enableDynamicPayloads();
  radio.setAutoAck(1);
  radio.setRetries(15,15);
  radio.setDataRate(RF24_250KBPS);
  radio.setPALevel(RF24_PA_MAX);
  radio.setChannel(76);
  radio.setCRCLength(RF24_CRC_16);
  radio.openReadingPipe(1,0xF0F0F0F0E1LL);
  radio.stopListening();
  radio.startListening();
}

void blink_error(){
  int m;
  for (m=0; m<200; m++)
  {
    digitalWrite(LED,HIGH);
    delay(200);
    digitalWrite(LED,LOW);
    delay(200);
  }
}

// Turn LED on if WiFiManager goes in Captive Portal Mode
void configModeCallback (WiFiManager *myWiFiManager) {
  digitalWrite(LED, HIGH);
}
```

closed 2016-10-18 by Avamander