nRF24/RF24 #298: Getting pipe number in Python

filipek92, 2016-10-17:
Hello,

I have a question, how can I get pipe number, where the received mesage comes from?

I now, how to do it in C++ (bool available (uint8_t *pipe_num)), but what about python?

Thanks for your work, your library is excelent 👍

djgraff209, 2016-10-18:
I just look a look @ this myself because I was interested.

The RF24 python wrapper defines this function:

```
bp::tuple available_wrap(RF24& ref)
{
    bool result;
    uint8_t pipe;

    result = ref.available(&pipe);
    return bp::make_tuple(result, pipe);
}
```

Based on that - under python, I would surmise that you get back a tuple of the result (booleanish) and the pipe number.

PaoloElMarinelo, 2016-12-05:
Hello,
I'm looking for the same thing but I a new python programmer so I didn't understand what kind of sentence I have to write in my python code to have the pipe number.
Could you help me ?
Thanks a lot.

filipek92, 2016-12-05:
Sorry, I didn't had a time to test it yet, so I can not say if the solution works. But I think   returned value is bolean, not tuple in python.... Maybe you can test, what is output of function rf24.available() in python3

PaoloElMarinelo, 2016-12-21:
I try that :
            if radio.available():
                pipe = radio.available_pipe()
                while radio.available():
                    len = radio.getDynamicPayloadSize()
                    temperature = radio.read(len)
                    print('got response from pipe {} size {} value {}'.format(pipe, len, temperature))

The response  for pipe is a tuple : `(True,1)`

closed 2019-05-10 by TMRh20