nRF24/RF24 #299: error compiling with static library on Raspbian

Slartitorto, 2016-10-21:
Hi all,
I want to compile with static library for to put the binary on another RaspberryPi; so I've inserted " -static" in the gcc command:
g++ -Ofast -mfpu=vfp -mfloat-abi=hard -march=armv6zk -mtune=arm1176jzf-s -Wall -static -I../ -lrf24-bcm myprogram.cpp -o myprogram
but make responds:
/usr/bin/ld: cannot find -lrf24-bcm
What may be the mistake?

TMRh20, 2016-10-25:
``` -lrf24-bcm ```

It is now just ``` -lrf24 ``` if using latest updates.

Slartitorto, 2016-10-25:
Same problem.
```
# uname -a
Linux raspberrypi 4.1.19-v7+ #858 SMP Tue Mar 15 15:56:00 GMT 2016 armv7l GNU/Linux
```

I've upgraded Linux with apt-get upgrade and reboot, then reinstalled RF24-master (latest version); after ./configure
```
# make
arm-linux-gnueabihf-g++ -fPIC -march=armv7-a -mtune=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -c RF24.cpp
cc1plus: error: bad value (cortex-a7) for -mtune switch
Makefile:40: set di istruzioni per l'obiettivo "RF24.o" non riuscito
make: *** [RF24.o] Errore 1

```
Then, i removed -mtune switch from Makefile, so it compiled. Make, make install, went fine.
Then i compiled my program:
```
# arm-linux-gnueabihf-g++ -march=armv7-a  -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib BARS1_receiver.cpp -lrf24 -lconfig -o BARS1_receiver
#
```
went fine without -static flag, but not with -static flag:
```
arm-linux-gnueabihf-g++ -march=armv7-a  -mfpu=neon-vfpv4 -mfloat-abi=hard -Ofast -Wall -static -pthread  -I/usr/local/include/RF24/.. -I.. -L/usr/local/lib BARS1_receiver.cpp -lrf24 -lconfig -o BARS1_receiver
/usr/bin/ld: impossibile trovare -lrf24
collect2: ld returned 1 exit status
#

```

TMRh20, 2016-10-26:
The following linkns should help:

Dynamic vs Static libraries: https://ubuntuforums.org/showthread.php?t=1725448

RF24 Linux cross-compilation: http://tmrh20.github.io/RF24/CrossCompile.html

closed 2016-11-10 by TMRh20