nRF24/RF24 #300: Broken/Inconsistent output of RPi scanner example

campenr, 2016-11-06:
The scanner example for the RPi found under ``examples_linux/extra`` produces a broken output compared to the output of the Ardunio version of the same example.

Specifically, the RPi version, rather than refreshing signal strength in each channel after each loop, instead increments the values as the previous values are never cleared (the Ardunio version clears after 100 passes, and only displays the sum of those 100 passes, not each single pass).

I believe this behavior of the RPi example is unintentional as the function for clearing the previous values is inside an if whose conditions can never be met, and that the Arduino versions behavior is the desired behavior.

campenr, 2016-11-06:
PR to fix this issue in #301.

TMRh20, 2016-11-07:
> Specifically, the RPi version, rather than refreshing signal strength in each channel after each loop, instead increments the values as the previous values are never cleared

Technically this is intentional by design. The reasoning is that the RPi has enough memory to maintain the values, and IMHO it is easier to have the CPU keep count of active carriers found per channel than having to go through the results manually.

Bad idea?

campenr, 2016-11-07:
Hmmm, I see what you mean. 

My main problem was that in it's current implementation the RPi example is inconsistent with the matching Arduino example which as someone new to this library confused me as I was under the impression that the examples should work the same.

I also found with the current RPi scanner example that channels that are perfectly usable with the gettingstarted example (for example), but occasionally have some signal (likely overflow from neighboring channels) would become saturated in the output which I found misleading.

My personal feeling is that for the matching RPi and Arduino examples they should produce the output. I believe doing otherwise is confusing. However, perhaps there could be an additional RPi only example that takes advantage of the behavior you describe, but named something different to avoid confusion?

TMRh20, 2016-11-07:
Hmm, looking closer at this I'm agreeing with you more and more.

It seems that the nrf24l01+ chips detect RPD in two different ways
1. 'Latch' RPD/Carrier when payloads are received
2. 'Latch' RPD/Carrier on ESB timeout or stopListening();

So it seems that by calling stopListening() before testCarrier(), it forces the RPD to 'latch' even if payload was not received, which is probably better behavior for a scanner.

In this case, I would maybe also suggest changing the Arduino example to match the behavior more closely by using the same process.

Here is the (confusing) explanation of it in the datasheet:
```
6.4 Received Power Detector measurements
Received Power Detector (RPD) ...  triggers at received power levels above -64
dBm ... 
The RPD can be read out at any time while nRF24L01+ is in receive mode. This offers a snapshot of the
current received power level in the channel. The RPD status is latched when a valid packet is received
which then indicates signal strength from your own transmitter. If no packets are received the RPD is
latched at the end of a receive period as a result of host MCU setting CE low or RX time out controlled by
Enhanced ShockBurst™.
The status of RPD is correct when RX mode is enabled and after a wait time of Tstby2a +Tdelay_AGC=
130us + 40us. The RX gain varies over temperature which means that the RPD threshold also varies over
temperature. The RPD threshold value is reduced by - 5dB at T = -40°C and increased by + 5dB at 85°C.
```

campenr, 2016-11-07:
So when I update the RPi code I had intended it to match the Arduino code, not noticing that the Arduino code doesn't call ``radio.stopListening():`` until after ``radio.testCarrier`` so that is my mistake. I don't understand the intricacies of RF to know which is better, but I definitely agree that whatever is used it should be the same across equivalent examples.

campenr, 2016-11-07:
Actually, it turns out I based the new RPi code on the scanner program in ``examples/scanner/output/scanner.cpp`` not ``examples/scanner/scanner.ino``. I knew I got that specific order from somewhere...

campenr, 2016-11-07:
So reading a bit more I agree, it makes sense to stop listening before checking for a carrier.

I have updated the Arduino scanner sketch so that it has this same behavior as the new RPi scanner program and committed/pushed to my existing PR #301 .

closed 2016-11-07 by TMRh20