nRF24/RF24 #302: Rpi 3 problems to communicate.

sbongain, 2016-11-10:
Hi, im having a lot of problems using the RPi3 and the nFR24l01+ module. 
i have no communication, i already enabled the spi, connected the pins to GPIO21 and CE0, no wiring problem info, changed the FR24 radio(21,0), and no answer. i've tried installing the libraries using git clone and using the "installer.sh"... but nothing.

the RPi3 is up to date (uname -a)
Linux Rpi 4.4.30-v7+ #919 SMP Tue Nov 1 16:57:28 GMT 2016 armv7l GNU/Linux

when i run the GettingStarted script i got this in the console, 
```
RF24/examples/GettingStarted/
================ SPI Configuration ================
CSN Pin  	= CE0 (PI Hardware Driven)
CE Pin  	= Custom GPIO21
Clock Speed	= 8 Mhz
================ NRF Configuration ================
STATUS		= 0x0e RX_DR=0 TX_DS=0 MAX_RT=0 RX_P_NO=7 TX_FULL=0
RX_ADDR_P0-1	= 0x65646f4e31 0x65646f4e32
RX_ADDR_P2-5	= 0xc3 0xc4 0xc5 0xc6
TX_ADDR		= 0x65646f4e31
RX_PW_P0-6	= 0x20 0x20 0x00 0x00 0x00 0x00
EN_AA		= 0x3f
EN_RXADDR	= 0x02
RF_CH		= 0x4c
RF_SETUP	= 0x07
CONFIG		= 0x0e
DYNPD/FEATURE	= 0x00 0x00
Data Rate	= 1MBPS
Model		= nRF24L01+
CRC Length	= 16 bits
PA Power	= PA_MAX

 ************ Role Setup ***********
Choose a role: Enter 0 for pong_back, 1 for ping_out (CTRL+C to exit) 
>1
Role: Ping Out, starting transmission 

Now sending...
failed.
Failed, response timed out.
Now sending...
failed.
Failed, response timed out.
Now sending...
failed.
Failed, response timed out.
Now sending...
failed.
Failed, response timed out.
```

and testing the same script in arduino mini pro,

```
*** PRESS 'T' to begin transmitting to the other node
Sent response 4294967295
*** CHANGING TO TRANSMIT ROLE -- PRESS 'R' TO SWITCH BACK
Now sending
Sent 17023100, Got response 0, Round-trip delay 14851184 microseconds
Now sending
Sent 32876312, Got response 0, Round-trip delay 14911536 microseconds
Now sending
Sent 48789892, Got response 0, Round-trip delay 15018528 microseconds
Now sending
Sent 64810468, Got response 0, Round-trip delay 14893532 microseconds
Now sending
Sent 80706048, Got response 0, Round-trip delay 15583528 microseconds
Now sending
Sent 97291624, Got response 0, Round-trip delay 15018988 microseconds
Now sending
Sent 113312648, Got response 0, Round-trip delay 15384100 microseconds
Now sending
Sent 129698844, Got response 0, Round-trip delay 15908152 microseconds
Now sending
Sent 146609084, Got response 0, Round-trip delay 15758620 microseconds
```
help plz....

thx!

Avamander, 2016-11-10:
You only have one problem, rPi doesn't communicate with the nRF. One thing
that has fixed the problem for most is a full reinstall on the rPi.

sbongain, 2016-11-15:
i have reinstalled raspbian in a new microSD, updated the RPi and raspbian, installed the FR24 again using the installer.sh and nothing happens. i'm having the same output when im running examples_linux/gettingstarted  "Now sending... failed. Failed, response timed out."...

TMRh20, 2016-11-15:
From the above output it looks like the arduino may be the issue.

One common mistake tho, the RPi pin listed as CE0 in Pinouts is the CS pin for the radios, with radio ce using any other pin.

sbongain, 2016-11-16:
oh, you are right @TMRh20 ... the problem is the arduino, the RPi is working fine, i changed my arduino mini pro for an arduino uno and now is working fine... thx @TMRh20 and @Avamander for your help and time.

closed 2019-05-10 by TMRh20