nRF24/RF24 #305: RF24 + wiringPi = 100% CPU load

aurelian17, 2016-11-20:
I was trying to use in the same project wiringPi for controlling some GPIO's and RF24 + RF24Network + RF24Mesh in order to communicate with some arduino modules.
By configuring in the same process wiringPi and RF24 I always get 100% CPU load on raspberrypi2 platform.
Is it any chance to have RF24 configured via wiringPi?

TMRh20, 2016-11-20:
I'm wondering if this is a rf24 code issue or other.

Typically you need a delay of at least 2ms in your code to keep CPU usage down and go easy on the radio. Else use interrupts. The radio will receive data on its own if RPi is sleeping too. Else use interrupts.

aurelian17, 2016-11-20:
The RF24 loop has a 250ms delay, because increasing this value means that the arduino modules get in trouble when trying to connect to the network.

Might be related to the fact that I'm trying to configure the wiringPi with wiringPiSetupGpio() function?

How hard would be to configure the SPI for RF24 via wiringPi ?

TMRh20, 2016-11-20:
Well there is a [portability template](http://tmrh20.github.io/RF24/Portability.html) but portability needs a bit of a review and cleanup. ( I wish I was a better programmer when I started this all )

In any case, one could just steal the RPI template(s) and update for wiringPi, and I think update the configure script. Feel free to open an issue if you are not able to do it, but it will just get put on the list after the other ones for now.

aurelian17, 2016-11-20:
I'll try to do that, if I'll have problems I'll post here. Is this ok for you? When everything is finished I'll open a pull request.

TMRh20, 2016-11-21:
Yup, or send me an email at gmail, user tmrh20

aurelian17, 2017-04-12:
I finished the implementation for this. I tested with RPi and one arduino. I don't know how to proceed in order to submit the changes here. I forked this repo and made the implementation in that git.

Oitzu, 2017-04-12:
Open a pull request with the issue number referenced, so that a user can review it.

aurelian17, 2017-04-12:
I'm trying to push my branch to remote and I get an error:

git push origin WiringPi_Driver_#305 
ERROR: Permission to TMRh20/RF24.git denied to aurelian17.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

Oitzu, 2017-04-12:
You need to push your commits to your own fork and then open a pull request to the tmrh20 fork on the github website. You can't push directly to the tmrh20 fork.

aurelian17, 2017-04-12:
Thanks @Oitzu for your support. It's first time for me submitting one change to other repository. Didn't know about this procedure.

closed 2019-05-10 by TMRh20