nRF24/RF24 #306: SPIDEV not working on raspbian

plasticassius, 2016-11-21:
I have an issue similar to #138. My radio works fine with the RPi driver; however, not with SPIDEV.

`
lsmod | grep spi
spi_bcm2835

ls -l /dev/spi*
crw-rw---- 1 root spi 153, 0 Nov 21 21:17 /dev/spidev0.0
crw-rw---- 1 root spi 153, 1 Nov 21 21:17 /dev/spidev0.1

 cat /proc/cpuinfo
processor	: 0
model name	: ARMv6-compatible processor rev 7 (v6l)
BogoMIPS	: 697.95
Features	: half thumb fastmult vfp edsp java tls 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xb76
CPU revision	: 7

Hardware	: BCM2708
Revision	: 000d
Serial		:

uname -a
Linux raspberrypi 4.4.23+ #913 Tue Oct 4 13:56:56 BST 2016 armv6l GNU/Linux
`
Since testing with spidev_test.c passes, I dug a bit deeper and found that my csn pin is stuck high. I checked the other csn pin and it is the same. Neither does it matter entry in the /dev directory I use.

The code in spi.cpp has errors in both the uncommented and commented versions since it leaves parts of the ioc uninitialized. It must have worked only by accident. Here are my fixes:

`
uint8_t SPI::transfer(uint8_t tx)
{
    pthread_mutex_lock (&spiMutex);
	int ret;

	struct spi_ioc_transfer tr;
    memset(&tr, 0, sizeof(tr));
	tr.tx_buf = (unsigned long)&tx;
	tr.rx_buf = (unsigned long)&tx;
	tr.len = sizeof(tr);
    tr.speed_hz = this->speed,	
	tr.delay_usecs = 0;
	tr.bits_per_word = this->bits;
    tr.cs_change = 1;

	ret = ioctl(this->fd, SPI_IOC_MESSAGE(1), &tr);
	if (ret < 1)
	{
        pthread_mutex_unlock (&spiMutex);
		perror("can't send spi message");
		abort();		
	}

    pthread_mutex_unlock (&spiMutex);
    return tx;
}

//void bcm2835_spi_transfernb(char* tbuf, char* rbuf, uint32_t len)
void SPI::transfernb(char* tbuf, char* rbuf, uint32_t len)
{
	pthread_mutex_lock (&spiMutex);
	int ret;

	struct spi_ioc_transfer tr;
    memset(&tr, 0, sizeof(tr));
    tr.tx_buf = (unsigned long)tbuf;
    tr.rx_buf = (unsigned long)rbuf;
    tr.len = len;
    tr.speed_hz = this->speed;
    tr.delay_usecs = 0;
    tr.bits_per_word = this->bits;
    tr.cs_change=1;

	ret = ioctl(this->fd, SPI_IOC_MESSAGE(1), &tr);
	if (ret < 1)
	{
        pthread_mutex_unlock (&spiMutex);
		perror("can't send spi message");
		abort();		
	}
    pthread_mutex_unlock (&spiMutex);
}
`
I also tried combining modes SPI_CS_HIGH and SPI_NO_CS and eliminating tr.cs_change = 1. Nothing made CS work, so I think the problem must be in the linux drivers themselves.

TMRh20, 2016-11-22:
I noticed with RPi that if I load up RF24 with the BCM driver, then try to use SPIDEV, it won't work.  
  
I'm assuming it is because the BCM driver puts the GPIO & SPI pins into a state that make them inoperable with SPIDEV until a reboot or (i think) defaulting the pins to their original state. Haven't looked into it a lot. 

So my best bet is that it does have something to do with the linux drivers, but they should be able to work as long as nothing else is interfering with the default pin state(s).

TMRh20, 2016-11-22:
Also, I'll have to look at the changes to SPI, things like ```tr.len = 1``` do seem way off.

plasticassius, 2016-11-22:
Oops, my mistake, tr.len = 1 is correct. I meant to replace it with tr.len = sizeof(tx) which is only a cosmetic change that gives the buffer size. I'll run some more tests to see if this changes anything. The problem I was trying to fix is in the initializer to tr.

In part, spidev_test.c has `struct spi_ioc_transfer tr = { .tx_buf = (unsigned long)tx, };` Apparently, this type of initializer is legal in C but not C++. It is by no means the same as `struct spi_ioc_transfer tr = { tr.tx_buf = (unsigned long)tx, };` which looks like a fix for the syntax error you get in C++, but completely changes what the code does. I don't think a C++ initializer can be used here since differences in the `struct spi_ioc_transfer` declaration in the distribution's header would break it.

In spidev.h it states we must zero-initialize the structure: `struct spi_ioc_transfer tr; memset(&tr, 0, sizeof(tr));` Then we can set the members we need to. The members we don't set will be 0, eliminating anything uninitialized.

I noticed that after I use the BCM driver, occasionally the SPIDEV driver works, but only for one transfer.

TMRh20, 2016-11-22:
Thanks for all the info, though I'm not sure what to suggest for the remaining issue.

plasticassius, 2016-11-23:
The following klunky thing actually works. However, at this point I'm inclined to avoid the SPI driver entirely.

`uint8_t SPI::transfer(uint8_t tx)
{
	pthread_mutex_lock (&spiMutex);
	int ret;

	digitalWrite(8,LOW);

	struct spi_ioc_transfer tr;
    memset(&tr, 0, sizeof(tr));
	tr.tx_buf = (unsigned long)&tx;
	tr.rx_buf = (unsigned long)&tx;
	tr.len = sizeof(tx);
    tr.speed_hz = this->speed,	
	tr.delay_usecs = 0;
	tr.bits_per_word = this->bits;
    tr.cs_change = 1;

	ret = ioctl(this->fd, SPI_IOC_MESSAGE(1), &tr);
	if (ret < 1)
	{
        pthread_mutex_unlock (&spiMutex);
		perror("can't send spi message");
		abort();		
	}

	digitalWrite(8,HIGH);
	pthread_mutex_unlock (&spiMutex);
	return tx;
}

//void bcm2835_spi_transfernb(char* tbuf, char* rbuf, uint32_t len)
void SPI::transfernb(char* tbuf, char* rbuf, uint32_t len)
{
	pthread_mutex_lock (&spiMutex);
	digitalWrite(8,LOW);

	int ret;

	struct spi_ioc_transfer tr;
    memset(&tr, 0, sizeof(tr));
    tr.tx_buf = (unsigned long)tbuf;
    tr.rx_buf = (unsigned long)rbuf;
    tr.len = len;
    tr.speed_hz = this->speed;
    tr.delay_usecs = 0;
    tr.bits_per_word = this->bits;
    tr.cs_change=1;

	ret = ioctl(this->fd, SPI_IOC_MESSAGE(1), &tr);
	if (ret < 1)
	{
        pthread_mutex_unlock (&spiMutex);
		perror("can't send spi message");
		abort();		
	}
	digitalWrite(8,HIGH);
    pthread_mutex_unlock (&spiMutex);
}
`

plasticassius, 2016-12-09:
The radio is working well for me with the above hack, thanks for the great library. I referred to this issue in [https://github.com/raspberrypi/linux/issues/1758](url)

pelwell, 2016-12-09:
See https://github.com/raspberrypi/linux/issues/1758 for my explanation of why I think the spidev and SPI drivers are fine and the fault is in the application. There are lots of people using spidev on RPi - we would know if there was a real problem.

plasticassius, 2016-12-09:
@pelwell thanks for pointing out the error. I created a pull request #315 with a proposed fix.

closed 2017-01-21 by plasticassius