nRF24/RF24 #307: nRF24L01 - RF24Mesh - Interference

TThanhXuan, 2016-11-22:
Hi everyone, I have a big problem with Interference.
I'm building a RF24Mesh network in my company, I'm testing the example codes with a Gateway and a Node.
I uploaded RF24Mesh_Example_Master_To_Nodes.ino to my Gateway and RF2Mesh_Example.ino to my Node. I use ESP8266 for my Gateway and Arduino Uno R3 for my Node.
Usually, after about 10 - 15 minutes, the network stops working (Node tries "Renewing address", Gateway stops receiving messages). I used "scanner.ino" in RF24 examples to check if there was interference and I found that there was interference on many channels. 
I'm trying to find a solution for this problem, recently I tried to change the channel manually in RF24Mesh_config.h or using mesh.setChannel(), but it's too inconvenient.
Could you please give me a solution to keep my network stable, avoid interference and it can run for a long long time ?
Thank you very much.

Avamander, 2016-11-22:
The issue is probably not interference but you can try decoupling the 3.3V
rail.

TThanhXuan, 2016-11-22:
@Avamander: I already have a 100uF decoupling capacitor on both gateway and node. After stopping for some minutes (sometimes 30 minutes), the network will run again.

TThanhXuan, 2016-11-22:
What I want to know is how to keep the network avoid interference.
Because I'm developing this project for my company, but It's very unstable :(

TMRh20, 2016-11-22:
Interference isn't directly addressed with RF24Mesh, but you can use channel hopping methods to avoid interference.

I've been thinking about a very simple channel hopping example using only 2 or 3 channels.

At the top of each sketch, add: 
```
uint8_t channels[] = {1,45,90};
uint8_t channelCounter = 0;
```

In RF24Mesh_Example.ino, copy this and paste over the existing code

```
      if ( ! mesh.checkConnection() ) {
        //refresh the network address
        Serial.println("Renewing Address");
        if(!mesh.renewAddress(3000)){
          mesh.setChannel(channels[channelCounter]);
          Serial.print("Channel: "); Serial.println(channels[channelCounter]);
          channelCounter = channelCounter >= 2 ? 0 : ++channelCounter;          
        }
      } else {
```

In Master_To_Nodes.ino, copy this and paste over the existing code:

```
uint32_t displayTimer = 0;

uint32_t nodeTimer = 0; 
void loop() {    
 
  // Call mesh.update to keep the network updated
  mesh.update();
 
  // In addition, keep the 'DHCP service' running on the master node so addresses will
  // be assigned to the sensor nodes
  mesh.DHCP();
 
  if(millis() - nodeTimer > 5000){
    nodeTimer = millis();
    mesh.setChannel(channels[channelCounter]);
    Serial.print("Channel: "); Serial.println(channels[channelCounter]);
    channelCounter = channelCounter >= 2 ? 0 : ++channelCounter; 
  }
  
  // Check for incoming data from the sensors
  if(network.available()){
    nodeTimer = millis();
```

Explanation:

The master node keeps track of how long since receiving a packet from any node, in this example if no packets received in 5 seconds, it starts listening on different channels, changing every 5 seconds.

The nodes running the mesh_example will change to a different channel each time a payload write fails, but the network tends more towards stability if using a timer based heartbeat instead of renewing on every failed write. I would recommend keeping the gateway timeout longer than the nodes, so the network can stabilize once a single node has connected.

This is about the most straightforward & simple channel hopping method for dealing with interference. I haven't tested it much, but it seems to work.

closed 2019-05-10 by TMRh20