nRF24/RF24 #309: Possibility to read ARC_CNT for maintenance/optimisation purpose (ENHANCEMENT)

pierre83, 2016-11-29:
Hello,
I modified your library RF24.cpp to include the possibility of reading ARC_CNT counter by creating the function:
uint8_t readARC_CNT(void)
{
  return ((read_register( OBSERVE_TX ) >> ARC_CNT) & 0b1111);
}
And RF24.h
  /**
   * Report retry count
   * for the previous operation
   */
  uint8_t readARC_CNT(void) ;

Do you think it could be useful to others including it ?
Thanks for your (very excellent) job, my meteo station is running since 3 years now, thanks to you !
Pierre

TMRh20, 2016-12-01:
Yeah, this might be worth adding. I'll check it out etc when able to, but it may take a while.

Avamander, 2019-10-25:
Potentially implemented or fixed by https://github.com/nRF24/RF24/pull/463

closed 2020-01-18 by Avamander