nRF24/RF24 #310: RF24.write

Okcolleen, 2016-12-01:
Hello, 

I am having some trouble with the speed of my program.  I use the RF24.write command in a loop with a digital output,  I want the digital output to output faster, but it seems to take 126mS to process the RF24.write command.  It's too slow for my application, how do I increase the speed of the write? I tried commenting out the while statement, but it didn't change the speed.  Could anyone help with this? 

Thank you.

TMRh20, 2016-12-02:
The max for a failed write command is somewhere around 56 to 85ms, so 126 has to be incorrect.

To lower write times however, either disable AA and/or use one of the other write commands ( [startWrite();](http://tmrh20.github.io/RF24/classRF24.html#aa27519fc289920094422033e0bbf8cf9) ) etc with interrupts which is demonstrated in some of the examples.

dimecoin, 2016-12-23:
Performance wise, is it better to use startWriteFast() or writeFast() ?

closed 2019-05-10 by TMRh20