nRF24/RF24 #317: Starting from "gettingstarted_call_response.cpp", How can I add other nodes?

sineverba, 2016-12-15:
First of all, thank you for your library. In a couple of minutes (really!) I did connect a PI2 and an Arduino.
In 30 minutes I did edit the "gettingstarted_call_response" to send a string from the PI to the Arduino and receive as ACK a string from Arduino. And I'm a PHP programmer, not a C++!

But, I'm going to buy another one Arduino and another one NRF module. First, I'm studying your code.

Take my example:

I want send the string "TMRH20 IS GREAT" from PI to Arduino 1.
I want send the string "GZ TMRH20 LIBRARY" from PI to Arduino 2.

I need "simply" edit the gettingstarted_call_response.cpp file doing this:

1) Increment (to max 5) the radioNumber on User Config section? But it's a boolean, so don't mean 0 == false and 1 == true?
2) Add to the const uint8_t pipes[][6] another one element (e.g. 3Node)?

3) Finally, I imagine that I need to edit this part:

```
if ( !radioNumber )    {
      radio.openWritingPipe(addresses[0]);
      radio.openReadingPipe(1,addresses[1]);
    }else{
      radio.openWritingPipe(addresses[1]);
      radio.openReadingPipe(1,addresses[0]);
    }
	radio.startListening();
```
But how select the receiver?

Thank you for your support.

Onmt39, 2016-12-30:
Have you had a look at the RF24Network library? Its easy enough to use to communicate between multiple nodes. You can read about how it works here: http://tmrh20.github.io/RF24Network/

TMRh20, 2016-12-30:
Maybe if you send the right kind of message, it will work properly. Something like "I am always good to people", and believe in yourself more than me? Or maybe you already do?

closed 2019-05-10 by TMRh20