nRF24/RF24 #320: Fix for dramatically increasing RF24Network transmission reliability @ 250Kbps

NeptuneTheMystic, 2016-12-20:
I've been poking and prodding this problem (and the nRF24L01+ datasheet) for over a week and cannot fathom why this solution works, so I'm submitting this information here in hopes that someone can shed some light.

This involves the RF24Network, but considering the fix is to the RF24.cpp file from this library, I figured I'd post it here.

Using two Arduino Unos connected to the PCB-antenna version of the Chinese clones of the nRF24L01+ modules.  The modules came from the same seller in the same shipment and appear identical.  Using 100uF electrolytic caps across the radio module GND & power pins.  Voltage provided by the onboard 3.3V regulator from the Arduino boards, which are connected via USB to two different computers in the same room.

Using the RFNetwork example sketches helloworld_tx on one and helloworld_rx on the other.  Using dynamic payloads and disabling fragmentation.  The only mods to the sketches are `radio.setDataRate(RF24_250KBPS);` right after `radio.Begin()` in both sketches.

The percentage of a successful transmission is about 2% - or less.  It is more successful at 1Mbps and even moreso at 2Mbps.

I troubleshot the problem down to the requirement of a delay - of at least 850us, although it seemed right on the cusp - 5ms improved the transmission to 100% reliability, absolutely rock solid.

What kills me about this is where the delay has to be.  I don't get it at all.  Line 998, so:

```
   while( ! (read_register(FIFO_STATUS) & _BV(TX_EMPTY)) ){
      delay(5);      
      if( get_status() & _BV(MAX_RT)){

```
It won't work if the delay is moved above the while loop.  But without that delay in there, it will buzz through the while loop until the timeout is reached and ultimately fail - most of the time.  The size of the payload also affects the delay; while 850us might be right on the cusp for a payload of a couple of longs, if you add more to it, the delay needs to be longer.

It makes no sense, but it is absolutely replicable for me.

I could just leave the delay there and move on, but ... I can't!  I have to know why it needs to be there.  I've swam through the datasheet over and over and cannot find an explanation.  Every time I think I've discovered an error in the RF24 code, it doesn't actually pan out.  I can't see anything wrong!

Anyone?  Can anyone else replicate this at this data rate?  Is there something you'd have me try?  Could you give me any guesses as to what you _think_ the problem might be and I can chase it further?  Can you explain why the delay has to be at that specific location in the code?

Thank you for your attention.

TMRh20, 2016-12-20:
I'm getting really far behind on these issues, but for this I would guess/suggest two things initially if not already attempted:
1. Set pa level to RF24_PA_MIN or LOW or move devices further apart
2. Create a grounded shielding for each radio device using tinfoil and plastic wrap (was a prev issue about this for pa + lna modules)

NeptuneTheMystic, 2016-12-21:
Sigh... the PA level was indeed the issue.  I used RF24_PA_LOW and the problem went away.  Put it back to HIGH and the problem recurred (well, unless I put the delay back in).  I'm not sure what's happening in the "real world" with the higher power level requiring the delay, but regardless, I'm going to keep that in because then it ensures that the radios work at all power levels regardless of distance.

Thank you for your time.

closed 2016-12-21 by NeptuneTheMystic