nRF24/RF24 #321: Does packet size matter to the FIFO Buffer for writeFast?

dimecoin, 2016-12-23:
According the datasheet, these have 3 x 32 byte TX FIFO buffers (and same for RX buffers).

With the function 'writeFast(&buf, len);'  the docs say that these will put it in the TX buffer.

How much does packet size matter here?  

For example, consider this code where we have 3 packets that are 9 bytes in size.

    for (int i=0; i<4; i++ ){
        writeFast(&buf, 9);
    }

let's assume the chip doesn't flush/send the data.  What is the buffer state after the above code?

Is it:

Buffer 1 has 27 bytes (3 packets x 9 bytes)  [ 32 buffer size - 27 data = 5 bytes free/unused]
Buffer 2 has 9 bytes (1 packet x 9 bytes)

Or does it automatically overflow?

Buffer 1 has 32 bytes [packet 4 overflows to next buffer]
Buffer 2 has 4 bytes

Or another way to phrase this question, is it better to size your data in factors of 32 to optimize buffer usage?

wmarkow, 2017-08-21:
>For example, consider this code where we have 3 packets that are 9 bytes in size.
> ```
> for (int i=0; i<4; i++ ){
>    writeFast(&buf, 9);
> }
> ```
I think this code will try to send **four** packets (not three).

From the **Figure 27. FIFO (RX and TX) block diagram** of [this](https://www.sparkfun.com/datasheets/Components/SMD/nRF24L01Pluss_Preliminary_Product_Specification_v1_0.pdf) manual I would say that the TX FIFO can hold up to three packets (no matter of the packet size). Your code above will write the first three packets to the chip's internal TX FIFO and the FIFO will be full. In the meantime the chip will start the transmission. At the fourth packet the **writeFast** method will wait (and block your program) until TX FIFO will not be full.

>Or another way to phrase this question, is it better to size your data in factors of 32 to optimize buffer usage?

If I'm correct about TX FIFO (my words above), then there is no way to optimize the buffer usage as the packet size is irrelevant.

TMRh20, 2017-08-21:
I believe the packet size does not matter, the fifo will hold three separate payloads and write fast will block when full.

closed 2019-05-10 by TMRh20