nRF24/RF24 #323: Why is flush_rx() in startListening() commented out?

soligen2010, 2017-01-04:
I see that flush_rx() was commented out in startListening().  I am curious as to why because I think I may need to flush the RX buffer in some cases just to be safe. (I am doing 80 channel FHSS with a 3ms hop interval)  I am still on the fence if I need to flush the RX buffer or not, but doesn't seem to hurt me if I do, so I am wondering if there have been issues that caused this to to be commented out.  I would like to do one of the following:

1. put flush_rx back into startListening()
2. overload startListening() with a flag that gives the option of having the rx buffer flushed
3. make flush_rx() public

Thoughts?  My leanings are to #3 as it is simplest and provides full flexibility.

BTW the issue I am having is that sometimes things get into a mode where only about half the packets are received - few enough to be an issue but still enough to keep the receiver from going to re-sync logic.  It is hard to tell exactly what is happening as this situation is intermittent.  One theory is that the timing sync is very borderline because a packet sat in the buffer too long before the the program realized it was there. This only happens sometimes (on start up), but of the unit starts up and is receiving 99  to 100% packets, then everything works great - it just gets into this weird mode sometimes at start up, and some transceivers seem to no do this at all while others do it more fairly regularly. (All transceivers tested on the same test harness).  I would rather flush the buffer and (hopefully) cause a re-sync to get good timing than let this odd situation persist.

wmarkow, 2017-03-17:
Take a look at this specific commit which comments the **flush_rx()**:

commit 1d1801500dc6135eede3921008c4f3d381a80252
Author: TMRh20
Date:   Tue Jul 29 00:20:22 2014 -0600

    Modify write,startWrite, remove rx buffer flushes
    
    Modified write() to use startFastWrite, then toggle CE after completion
    to remove need for a delay on some boards.
    Added delay to startWrite for non-Arduino boards.
    Removed receive buffer flushes. This prevents an issue in the
    RF24Network library, where payloads would be received, but flushed when
    a transmission takes place prior to reading.
    Should address issues #7 and #11

sherikapotein, 2017-03-30:
I had similar requirements. A slight modification to the library did it:
Add following function declaration to the RF24.H under public:
`uint8_t flushTx(void);`

and corresponding definition in  RF24.cpp
```
uint8_t RF24::flushTx(void)
{
	uint8_t result = flush_tx();
	return result;
}
```

soligen2010, 2017-03-30:
flush_tx is already a public method, so I don't think you really needed to do this.  flush_rx on the other hand is private.  I had considered putting in a pull request to make flush_rx public as well, but I ended up not using it as public so I held off on the change.  Do you need a public flush_rx?

sherikapotein, 2017-03-31:
Ahh.. must have been .. I actually required the flush_rx() to be made accessible, but probably ended up implementing flush_tx() too.. My scenario did require a public flush rx with many NRFs transmitting on the same pipe and during every predefined/fixed window of time only one pair was to be active. I tried using stop listening for other nodes but it wasn't as successful than flushing the rx buffer until an active window period is reached.

soligen2010, 2017-05-16:
Pull request was merged making flush_rx public

closed 2017-05-16 by soligen2010