nRF24/RF24 #325: startListening or stopListening breaks analogWrite on Arduino

HorstBaerbel, 2017-01-12:
I use an Arduino Pro Mini 3.3V 8MHz (328P) here, with a nRF24L01+ module attached to pins 7(CE), 10(CS) and 11-13 (SPI). There is a led with resistor attached to pin 9 which should be PWM-capable. I set the brightness of the led using analogWrite(), which works fine as long as there's no startListening() or stopListening() in my sketch. When I add one of those to the sketch the led does not blink anymore and the serial port one transmits gibberish. If I use a digitalWrite() instead of the analogWrite() it works fine too, so the electrical circuit shouldn't be the culprit.
My Arduino IDE version is 1.6.12 and the AVR board library 1.6.17 is installed.
Here is the (rather long) complete code. You can try the effect by uncommenting line 263 or 264:
```
//
// Arduino gamepads using the nRF24L01+ transceiver chips. In my example the NRF24duino board is used.
// The power draw of an Arduino Pro Mini 3.3V plus a nRF24L01+ transceiver is ~8-10mA.
// So theoretically, it should run ~300-400h from four AAA batteries (not counting self-discharge ;).
// At the moment this can read ~10 buttons, which is about the limit of pins available
// on the Arduino. This could be expanded with a shift register or MCP23017 or the like...
// Pins have pull-ups activated and buttons are LOW active. One pin acts as an "auto fire" toggle. 
// When LOW, some buttons (A, B, X, Y) will toggle their on/off state with ~10Hz, 
// as long as they're pressed down.
// The board connects A7 to a 6.8k / 10k voltage divider to measure battery voltage.
// An LED on pin 9 controlled via PWM will show different states of the board. The LED will:
// Pulse slowly: When the gamepad is looking for a receiver.
// Be steady on: When the gamepad has connected to a receiver.
// Blink rapidly: The gamepad battery is running low.

#include <SPI.h>
#include <RF24.h>

//-----serial------------------------------------------------------------

#define SERIAL_OUTPUT

#ifdef SERIAL_OUTPUT
    // this is needed for radio.printDetails()
    // don't forget call printf_begin() after Serial.begin()...
    #include <printf.h>

    void printBits(uint16_t myByte) {
        for(uint16_t mask = 0x8000; mask; mask >>= 1) {
            Serial.print(mask & myByte ? '1' : '0');
        }
    }
#endif

//-----power------------------------------------------------------------

// The ADC pin we measure battery power on
#define PIN_POWER_MEASURE A7
// Battery voltage should be 4-5V for alkaline batteries and 4-4.8V for rechargable batteries.
// The voltage divider will bring this into a range of 2.4-3V resp. ADC values of ~745-930.
#define BATTERY_LOW 745

bool isPowerOk()
{
    // we measure the voltage on PIN_POWER_MEASURE.
    return analogRead(PIN_POWER_MEASURE) > BATTERY_LOW;
}

void setupPower()
{
    pinMode(PIN_POWER_MEASURE, INPUT);
}

//-----gamepad state------------------------------------------------------------
bool isConnected = false;
bool isBatteryLow = false;

#define PIN_STATE_LED 9
// we update the LED state at ~20Hz
#define LED_UPDATE_INTERVAL (1000/20)
#define LED_BRIGHTNESS 128
#define LED_PULSE_INTERVAL (1000*1)
#define LED_BLINK_INTERVAL (1000/6)
uint32_t lastLedUpdateMillis = 0;
uint32_t ledUpdateRemainingMillis = LED_PULSE_INTERVAL;
int8_t stateLedFadeDirection = 1;
[... 222 lines cut ...]
    // set up serial port and print radio and button state
    Serial.begin(115200);
    Serial.println(F("Started"));
    Serial.println(F("Radio state: "));
#endif
    setupRadio();
}

void loop() {
    updateGamepadState();
    // check if we need to check the button state again
    if ((millis() - lastButtonUpdateMillis) >= BUTTON_UPDATE_INTERVAL) {
        lastButtonUpdateMillis = millis();
        // yes. check buttons
        const uint16_t newState = getButtonState();
        // check if the state changed or we need to send a heartbeat
        if (newState != buttonState || (millis() - lastMessageSentMillis) >= HEARTBEAT_INTERVAL) {
            lastMessageSentMillis = millis();
            // replace button state with new one and send it
            buttonState = newState;
            sendState(buttonState);
        }
    }
}
```

HorstBaerbel, 2017-01-12:
A minimal example sadly does NOT show this problem:
```
#include "RF24.h"

RF24 radio(7, 10);
uint16_t data = 0;

void setup() {
  pinMode(9, OUTPUT);
  Serial.begin(115200);
  Serial.println(F("Started"));
  radio.begin();
  radio.setPALevel(RF24_PA_LOW);
  radio.setChannel(123);
  radio.setDataRate(RF24_250KBPS);
  radio.setAutoAck(true);
  radio.setRetries(2, 8);
  radio.setCRCLength(RF24_CRC_8);
  radio.setPayloadSize(2);
  radio.openWritingPipe(0xF0F0F0F0D2LL);
  radio.openReadingPipe(1, 0x3A3A3A3AD2LL);
}

void loop() {
  analogWrite(9, data++);
  radio.stopListening();
  radio.write(&data, sizeof(data));
  radio.txStandBy();
  radio.startListening();
  delay(10);
  Serial.println(F("Blink"));
}
```
If the problem is in my code, where should I start looking?!

dimecoin, 2017-01-14:
Do your lines/wires touch?  I had a problem that if the line of my button switch [gnd] touch IRQ line it would "knock out" my rf24 device... 

These things are really sensitive to noise.  I know you heard it before, but you need clean lines and clean power.  Caps on gnd/vcc of RF24 help.

Avamander, 2017-01-14:
I'd rather say that the clones are really sensitive to noise.

HorstBaerbel, 2017-01-14:
I've already tried a 47uF cap on Arduino VCC/GND. It is definitely the analogWrite() + start-/stopListening(). A digitalWrite() + start-/stopListening() works just fine. I'm not sure how to attribute that to noise because the simple sketch works just fine. I was thinking the RF24 library sets something in the Arduino that breaks analogWrite() or vice-versa. [analogWrite() modifies a timer](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring_analog.c). Could that be a problem?

Avamander, 2017-01-14:
I don't remember the intricate details, but yes, it could mess up the SPI
communication.

HorstBaerbel, 2017-01-14:
I tried quickly toggling the LED (on/off) on pin 9 using digitalWrite() and it does not produce the problem, so I suspect that power/noise is not the issue. I can add as many start-/stopListening() as I want...

HorstBaerbel, 2017-01-14:
Problem solved. I did an analogRead() on A7 and digitalRead() on A6. Though these pins do exist on the 328P and on my board, they do not exist on the UNO (The board has the booloader flashed with the board set to UNO and I compile code in the Arduino IDE set to UNO). I guess something goes wrong when doing the analogWrite() I got no idea what though... :/

HorstBaerbel, 2017-01-14:
Thanks for the help y'all. The problem was in front of the screen again... ;/

closed 2017-01-14 by HorstBaerbel

brundle, 2017-03-17:
Any update/Info on the NRF Duino board?  I'm building a compact combination or NRF and Pro Micro for a hand-held remote and would love to eliminate some wires and area.  Reliability, range, etc.  Thanks.

HorstBaerbel, 2017-03-19:
The board works fine, but I don't have any figures on range. They can be driven from 5V via Vin and work just fine in the Arduino IDE and the RF24 lib (if you do it correctly, see above... ;). 
The seller was pretty cooperative too. The first set was delivered with the wrong bootloader and they quickly responded and sent another set with the correct bootloader.