nRF24/RF24 #327: Spidev lack of hardware support should not lead to a segmentation fault

razerraz, 2017-01-18:
On linux with spidev selected as access method, having non existing device lead to a segfault.
It can be a problem a least when using the library in a python thread, since it crash the entire process
Espected behaviour : having an error that can be handled via try/except method

plasticassius, 2017-01-18:
I take it that this is an offshoot of your issue with the Beagle Board related to pull #315. I can't reproduce this, however, I'm using Raspberry Pi and no Python. When the hardware doesn't exist, the library calls abort which exits the process when it is linked to C code, so I imagine it could do something fatal when linked to python. Is this different than the way it behaved before pull #315?

razerraz, 2017-01-18:
Clearly not related to #315. It occurs since I had began to use the library, 6 months ago...
Seems to be cpython related, since you don't have this issue using C directly

closed 2019-05-10 by TMRh20