nRF24/RF24 #329: pingpair_dyn.cpp:(.text.startup+0x378): undefined reference

Lan411, 2017-01-21:
Trying to compile the file but I received an error.
Please help
[image]

Avamander, 2017-01-21:
Did you modify it?

Lan411, 2017-01-21:
I did modify the other files

Gettingstarted, transfer and gettingstarted_callresponse.

I commented the sleep function as I got errors in all the files for it.
Not sure if this is the issue.

Lan411, 2017-01-21:
I redownloaded the library.
It all compiles now ...but I am getting a segmentation fault.
[image]

Coder-256, 2017-04-10:
Run it with `sudo`.

closed 2019-05-10 by TMRh20