nRF24/RF24 #341: TX Arduino UNO, RX ARduino DUE, struct variable

romik2206, 2017-03-10:
Hallo, I have small problem with sending struct variable. On side of transmitter I have...

_struct Sensor_data {byte ID; byte TX_Power; short Temp; float Pressure; byte Humidity; byte Battery; };
Sensor_data Sensor;_		

And I am sending it with...   _RF24.write(&Sensor, sizeof(Sensor));_

Serial debugging says me... **"Writting 10 bytes, 0 blanks."**   I think it is ok, 10 bytes is lenght of struct "Sensor".

On side of receiver I have...

_struct Sensor_data {byte ID; byte TX_Power; short Temp; float Pressure; byte Humidity; byte Battery; };
Sensor_data Sensor;_		

and I am reading it with...   _RF24.read(&Sensor, sizeof(Sensor));_

It perfectly works on another Arduino UNO or MEGA... AVR arch. Serial debugging says **"Reading 10 bytes, 0 blanks".**

It doesnt work on ARDUINO DUE, beacuse serial debugging says me... **"Reading 12 bytes, 0 blanks."**   

Problem is variable "Sensor.Pressure" beacuse it is float. The documentation says... **implementation of "float" is same on AVR (UNO, MEGA) and ARM (DUE) ... size of float is 4 bytes always.** 

Therefore the reading on RX side should be 10 bytes in both of cases (AVR, ARM) not 12 bytes as debugging says.

What is wrong? Where do I make mistake?

Thank you, best regards Roman

wmarkow, 2017-03-10:
The first feeling is that the sizeof for **short** and **float** may be different on different platforms/compilers (no matter what the documentation says). Could you just make a simple test and print out the sizeof for **short** and **float** separately on both platforms?

You can try to use **int_16t** though (or something similar) on both platforms but you will loose some accuracy for pressure then (but it depends on what you need).

romik2206, 2017-03-10:
Hi, ok I will try to test sizeof() of different types of variables.
I worked with documentation for Arduino. There is written... for example...
       AVR    ARM
int      2         4
short  2         2
float   4         4
double 4       8

But as you mentioned...

When structures are defined, the compiler is allowed to add paddings (spaces without actual data) so that members fall in address boundaries that are easier to access for the CPU.

_For example, on a 32-bit CPU, 32-bit members should start at addresses that are multiple of 4 bytes in order to be efficiently accessed (read and written). The following structure definition adds a 16-bit padding between both members, so that the second member falls in a proper address boundary:_

It should be a point of problem... Is it same for C and C++, I guess?
I will try it with attribute __packed__

Thx.

Avamander, 2017-03-10:
Use uint-s or int-s, they should be standard sized.

romik2206, 2017-03-10:
I think wmarkow gives me the correct answer... I will work with it...

closed 2017-03-10 by Avamander