nRF24/RF24 #342: Question: DigiSpark RF24 -is P5 available for use?

maurbeck, 2017-03-13:
I'm trying to read analog temperature values from P5 and transmit them using RF24. I was able to get everything running the the transmitter, but when I tried to read the TMP36 from P5 it stopped sending. Is the RF24 lib making use of P5 internally for something?

`int tempReading = analogRead(5);

  float voltage = tempReading * 5.0;
  voltage /= 1024.0;
  float tempC = (voltage - 0.5) * 100;
  float tempF = (tempC * 9.0 / 5.0) + 32.0;
  
  radio.stopListening(); // First, stop listening so we can talk.
  payload = tempF;
  radio.write( &payload, sizeof(unsigned long) );`

wmarkow, 2017-03-13:
DigiSpark uses ATtiny85, right? in ATtiny85 PB5 pin has four different alternate functons (like RESET for instance). Maybe it is something wrong with the PB5 pin configuration?

maurbeck, 2017-03-14:
Oh, do I need to configure PB5 for use as an analog input pin or something?

maurbeck, 2017-03-14:
Ha, I answered my own question. Of course I need to.

I needed "pinMode(Pin, INPUT);"

And also, PB5 is pin 0, not 5...

maurbeck, 2017-03-14:
Of course I need to set it as input with pinMode. And I figured out PB5 is pin 0 "pinMode(0, INPUT);"

But it seems like the reset functionality of PB5 is kicking in when I try to use it as an analog in.

wmarkow, 2017-03-14:
Actually I'm not so familiar with ATtiny85 but the documentation says that the RESET functionality is controlled by some **fuse-bit**:
>RESET: External Reset input is active low and enabled by unprogramming (“1”) the RSTDISBL Fuse. Pullup i activated and output driver and digital input are deactivated when the pin is used as the RESET pin.

You can check out the fuse bits in your ATtiny85 (I think you need some additional software to do that, I do not know if ArduinoIDE can do that) and disable the RESET functionality if needed.

maurbeck, 2017-03-14:
Thanks for the reference. I don't have enough pins to do a few other things I'm looking to do now, so I'm dropping this endeavor. 

Thanks again

closed 2017-03-14 by maurbeck

Avamander, 2017-03-14:
You can connect a RF24 using 3 pins.
https://www.hackster.io/arjun/nrf24l01-with-attiny85-3-pins-74a1f2