nRF24/RF24 #343: Receiver misses every 4th transmission

Brjodaro, 2017-03-14:
Hello!

I have a pair of nRF24L01+ chips/Teensys setup with a simple noack broadcast code. The transmitter sets a pin high when it has successfully sent a byte and the receiver sets a pin high when it has received it. I have a scope connected to these two pins to confirm that my broadcast is consistently being received. What I found, however, is that the receiver misses every fourth broadcast regardless of the time between transmissions. I've been stuck on this all day. Could anyone point me in the right direction? Thank you!

PS - I also found that if I substitute "writeFast" for "write" in the code below the transmitter sends nothing. What's that about?

Receiver code:
===========================================================

`#include <SPI.h>
#include "RF24.h"

#define INTERRUPT 1

RF24 radio(22,15);

const uint64_t pipes[2] = { 0xEEFAFDFDEELL, 0xEEFDFAF50DFLL };
static char sendByte[] = {0xAA};
char        receiveByte[100];

void setup(){
  pinMode(INTERRUPT, OUTPUT);
  Serial.begin(115200);
  radio.begin();

  radio.enableDynamicPayloads();
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_2MBPS);
  radio.setCRCLength(RF24_CRC_DISABLED);
  radio.setRetries(0,0);
  radio.setAutoAck(false);
  radio.enableDynamicAck();

  radio.openReadingPipe(1, pipes[1]);
  radio.openWritingPipe(pipes[0]);

  radio.powerUp() ;
  radio.startListening();
}

void loop(){

  //radio.write(sendByte, 1, true);

  digitalWrite(INTERRUPT, HIGH);
  delayMicroseconds(10);
  digitalWrite(INTERRUPT, LOW);

  while(!radio.available());
  
//Serial.print(radio.available());
//Serial.print("\t");
  while(radio.available()){ 
    radio.read(receiveByte, 1);
  }
  Serial.println(radio.available());
}`

Transmitter code: 
===========================================================

`#include <SPI.h>
#include "RF24.h"

#define INTERRUPT 1

RF24 radio(22,15);

const uint64_t pipes[2] = { 0xEEFAFDFDEELL, 0xEEFDFAF50DFLL };
static char sendByte[] = {0xAA};

void setup(){
  pinMode(INTERRUPT, OUTPUT);
  Serial.begin(115200);
  radio.begin();

  radio.enableDynamicPayloads();
  radio.setPALevel(RF24_PA_MAX);
  radio.setDataRate(RF24_2MBPS);
  radio.setCRCLength(RF24_CRC_DISABLED);
  radio.setRetries(0,0);
  //radio.setAutoAck(false);
  radio.enableDynamicAck();

  radio.openReadingPipe(1,pipes[0]);
  radio.openWritingPipe(pipes[1]);

  radio.powerUp() ;
  //radio.startListening();
}

void loop(){
  //Serial.println("here");
  //radio.stopListening();
  
  if(radio.write(sendByte, 1, true)){
  radio.flush_tx();
  //radio.startListening();
  digitalWrite(INTERRUPT, HIGH);
  delayMicroseconds(10);
  digitalWrite(INTERRUPT, LOW);
  }
  delayMicroseconds(16000);
}
`

wmarkow, 2017-03-14:
In your receiver code you have:
```
while(radio.available()){ 
    radio.read(receiveByte, 1);
  }
```
If I remember correctly RF24 has a small buffer for incoming packets. It could be, that if you have like two packets ready to read from nRF24 (from its internal buffer) then you will read them all in the loop but you will set the pin to high only once and you see only one spike on the scope. That would be worth to check. But another question is: why are the incoming packets stacked in the buffer queue? Maybe the transmitter sends to fast and receiver is too slow?

Brjodaro, 2017-03-14:
I was thinking about that, but at 60hz (delay(16000us)) the receiver should have long since processed the last byte sent by the transmitter and be ready for the next one. I did try serial printing the return value of radio.available() before and after the loop you mentioned and it always showed 1 before and 0 after.

wmarkow, 2017-03-14:
That's the point: if you put **radio.available** before loop then it may show 0 or 1. The **radio.available** after the loop will rather always show 0 because the loop just read all of the packets.
To be sure put the setting the pin to HIGH inside the loop after **radio.read(receiveByte, 1)**. Then you will see one spike per one packet. It may not solve the problem but at least will eliminate one possibility.

Brjodaro, 2017-03-14:
With the pin being set high inside the while loop, the scope actually shows two pulses on each reception about 100us apart. The first radio.read() in that loop gives the byte sent by the transmitter (0xAA) and the second is just zero. It still skips every fourth transmission.

wmarkow, 2017-03-16:
So what now? Why the receiver gets the 0x00 byte response? Maybe it is related to the dynamic payload? Maybe you could slow down the transmitter to send one packet per second and see what gets the receiver?

closed 2019-05-10 by TMRh20